thonny/thonny #1158: Thonny hangs on Tools, Options and Help, About (Windows)

retrosnob, 2020-04-15:
Where can I find error logs?

aivarannamaa, 2020-04-15:
The logs folder can be opened with "Tools => Open Thonny data folder". NB! The logs are currently overwritten each time you start Thonny.

I've seen that sometimes a dialog (eg. About) can end up behind the main window and this can effectively lock the UI. You can try whether you can reorder the windows by Alt-Tabbing to and away from Thonny.

retrosnob, 2020-04-24:
That was the problem. It was appearing off-screen. Thank you.

aivarannamaa, 2020-04-25:
Good to know! I'll leave it open, because this problem still needs a solution.

aivarannamaa, 2020-09-27:
Somwhat related to #1133

aivarannamaa, 2020-09-28:
Need to take #840 and https://bugs.python.org/issue34927 into account.

closed 2020-10-18 by aivarannamaa