thonny/thonny #1159: thonny startup error: no module named "pwd"

clarissa12193, 2020-04-15:
[image]

when i try to open thonny on my windows 10 laptop, i get the thonny startup error: no module named "pwd"
please help asap

aivarannamaa, 2020-04-15:
Seems to be related to #1146

Please check whether restarting your computer cures this problem!

clarissa12193, 2020-04-16:
it worked, thank you so much for the fast reply

closed 2020-04-16 by clarissa12193

aivarannamaa, 2020-04-16:
Do you have any ideas what could have caused this situation (of `USERNAME` environment variable not being set)? Were you using a recently created account or did you install or update something before this problem occurred?

clarissa12193, 2020-04-16:
No idea what had caused it, but I had bought a new laptop and installed
thonny on it, at first I could only open it if I ran the app as
administrator, but I couldn't open programs I'd have already made on thonny
as this error occurred, then I restarted it and now it opens as soon as I
click the app.
Thank you for your help