thonny/thonny #1162: Connect to several ESP boards via WebREPL (MicroPython)

LDmicroGitHub, 2020-04-16:
There is possible to connect to several ESP  boards via REPL through serial ports (for example COM7, COM8, etc.)  
Every board is configured WebREPL on different IP addresses (for example '192.168.4.7', '192.168.4.8').
(Not default WebREPL IP address192.168.4.1)  
In this case, we need a text field to edit the target WebREPL IP address.  
[image]

aivarannamaa, 2020-04-17:
When you select <WebREPL> the IP address and password entry boxes should appear below port combobox. You may need to resize the options window.

It looks like you don't have the newest Thonny version, because I've disabled WebREPL support for now because it is too unreliable.

LDmicroGitHub, 2020-04-17:
I try to use latest master branche with 
    def allow_webrepl(self):
        return True

LDmicroGitHub, 2020-04-17:
[image]  
After restarting the backend, an error occurs.  
AttributeError: 'function' object has no attribute 'put'  
[image]

aivarannamaa, 2020-04-17:
I guess this commit is to be blamed: https://github.com/thonny/thonny/commit/9cc673373e3579c7374f3203d5dd9885132ece96#diff-bb25fa39cfa7e96bf57039b80f3ee8b4

You should revert the changes in webrepl_connection.py

aivarannamaa, 2020-04-17:
... or better, just get rid of extra `.put`  on line 90

LDmicroGitHub, 2020-04-17:
After the changes in webrepl_connection.py  
in line 90 and 180  
>          #self._make_output_available.put(data, block=False)  
>          self._read_queue.put(data, block=False)  
and in line 180  
>          #self._make_output_available(data)  
>          self._read_queue.put(data)  
I get a message in the Shell window:  
> WebREPL connected  
> raw REPL; CTRL-B to exit  
>  
MicroPython device files are displayed in their window. It's Ok.

But when I press the 'Run current script' button I get an error
[micropython-backend.log](https://github.com/thonny/thonny/files/4492572/micropython-backend.log)

aivarannamaa, 2020-04-17:
I'll deal with it after I've finished with MicroPython communication refactoring (#1103).

LDmicroGitHub, 2020-04-17:
Thanks. I am ready to test a pre-release.

aivarannamaa, 2020-04-18:
Great! I'll let you know.

laukejas, 2020-10-22:
Hey, @aivarannamaa , what is the status on WebREPL support? I have tried enabling it in the esp\__init__.py and micropython\__init__.py, but it returns an error `ModuleNotFoundError: No module named 'websockets'`. Is there a workaround or something? We need WebREPL support real bad.

aivarannamaa, 2020-10-22:
@laukejas  Please try https://github.com/thonny/thonny/releases/tag/v3.3.0b2 There are some known problems, though.

If you run from source in a virtualenv, then you need to `pip install websockets`, because this is not a hard dependency.

laukejas, 2020-10-22:
@aivarannamaa Thank you, this is looking really great so far. What import issues are there to keep an eye out for?

aivarannamaa, 2020-10-22:
> What import issues are there to keep an eye out for?

No import issues (if you install `websockets`), but WebREPL has problems with long inputs (https://github.com/micropython/micropython/issues/2497). I have planned a work-around, but haven't got to it yet.

aivarannamaa, 2020-10-27:
WebREPL support will be much better in the next beta

closed 2020-10-27 by aivarannamaa