thonny/thonny #1165: Seperate Shells

EkremDincel, 2020-04-18:
I am making a socket application and I need to start both ``server.py`` and ``client.py```at a time, but there is only one Shell window in Thonny. Is there any way for running multiple files in seperate shells (like as we open multiple python files in separate windows)?

aivarannamaa, 2020-04-18:
Nope. Thonny is meant for beginners and for most users this feature would increase the complexity of the UI without bringing any benefits.

EkremDincel, 2020-04-20:
Can't we make this optional?

aivarannamaa, 2020-04-20:
I don't want to make it even optional, because it would require too much work.

But now it occurred to me that Thonny already has feature that can solve your case -- Run => Run current script in terminal. You can run one of your scripts with this and it leaves Thonny's shell for the other.

EkremDincel, 2020-04-20:
Yeah I know "Run current script in terminal" option but this don't works like shell after program exits, which is bad for debugging.

aivarannamaa, 2020-04-20:
Did you try "Tools => Options => Terminal => Present Python REPL after program ends" ?

cdlm, 2020-11-03:
Somewhat related —that would probably make more sense as a separate issue/feature request— we see confusion among novice students when we introduce I/O functions during the course. Because I/O happens in the same window as the REPL, students do not understand the difference between `return` and `print()`, and when one tries to explain, it doesn't help that the REPL shows barely any visible difference between a function that returns a value and one that prints it.

Of course there's a pedagogy angle to solving this, but it would help greatly if Thonny had a separate transcript window _à la Smalltalk_ where user program interactions happen. The REPL would then only serve to interact with the Python interpreter, and the Transcript with the user program.

If there's a REPL theme or configuration that can mimic the `In[n]` and `Out[n]` behavior of notebooks to differentiate effects from results, that would be a first step… I've quickly looked through Thonny's settings but haven't found something like that.

aivarannamaa, 2020-11-03:
@cdlm, if I understood you correctly, then "Run => Run current script in terminal" would give you the window which deals only with the IO of the user program.

cdlm, 2020-11-03:
Yes… in fact that's what we tried to guide the students towards during the lecture. IMHO that's at best a workaround, because it involves
- switching away from Thonny
- running in a separate process, so you can't try printing from the console and stuff appear in that terminal window
- not fixing the fact that outputs and results look virtually identical in the console

And also it fails to actually work on my machine, for some reason. It does launch Terminal.app (instead of iTerm2, but nevermind) but I only get a system shell prompt.

BTW I do understand a proper solution involves a custom REPL with separate redirectable program I/O; still I believe it has real teaching value.

aivarannamaa, 2020-11-03:
Actually Thonny has custom REPL, and one of the reasons for this was to distinguish between program IO and other interactions.

But IMO the current solution (indenting program IO and using different font) is sufficient. I've experimented with inverted colors, but the result was too noisy.

closed 2021-01-19 by aivarannamaa