thonny/thonny #1167: Don't force users to Stop/Restart before Run, if application terminates with OS call

ddekany, 2020-04-19:
If you run this:

```
import os
print("Hello world")
os._exit(0)
```

then Thonny prints "Backend terminated or disconnected. Use 'Stop/Restart' to restart." (so far fine... though it doesn't tell the exit code), but then Run is grayed out, and first you have to press Stop/Restart, and then Run (F5). Is this truly necessary, especially if the exit code was 0 (no error)? Why can't the user just run the application again?

Surely you ask, why would anyone exit with `os._exit(0)`. In my case, Pyxel, which is a thin Python wrapper around a C game engine, handles quitting itself, and apparently it just exits with OS call.

aivarannamaa, 2020-04-20:
I made the reconnection manual to avoid endless loop when the subprocess ended because of wrong interpreter (eg. Python 2).

I should make an exception for this when the process ends because of user code and when the exit code is 0. Also, it should show the exit code.

ddekany, 2020-04-20:
Or, even when the process exits with non-0, if then it only relaunched when Run is pressed, then there won't be an infinite loop.

aivarannamaa, 2020-04-20:
You're right!

aivarannamaa, 2023-08-19:
Fixed in version 4

closed 2023-08-19 by aivarannamaa