thonny/thonny #1168: pandas not working even though installed

clarissa12193, 2020-04-20:
when i try to run a pandas program, the following comes in the terminal
[image]

this is the program
[image]

aivarannamaa, 2020-04-20:
It seems to be problem with Windows build of latest Pandas: https://github.com/pandas-dev/pandas/issues/32857

As a workaround you should try going back to version 1.0.1. If you installed pandas with Thonny's pip GUI, then use "..." button for this:

[image]

clarissa12193, 2020-04-21:
> pandas-dev/pandas#32857

it worked, thank you so much for your help

closed 2020-04-21 by aivarannamaa