thonny/thonny #1172: Best way to install packages remotely/silently

AndrewBreyen, 2020-04-23:
Hello,

I am struggling to figure out a way to remotely deploy Thonny and packages. 

I was able to get the EXE to install fine remotely/silently by using this CMD script:
`"\path\to\thonny-3.2.7.exe" /VERYSILENT /NORESTART /DIR="C:\Program Files\Thonny" /LOG="C:\logs\thonny3.2.7.log"`

but now the requesting professor wants packages to be installed. The professor wants these packages installed:
[image]

Our network/user setup does not allow persistence in the user APPDATA directory, and we have tried having users manually install packages. This is tedious for our users, and doesn't work. 

Thonny must be installed to C:/Program Files in order to stay on the machines. 

Last year I believe there was a customize.py file that allowed us (very jankily) to point Thonny to the users network drive, where the user had previously copied the .thonny directory. This however appears to be missing from thonny v3. 

How should packages be remotely deployed silently, and across hundreds of computers?

Any suggestions?

Thank you.

AndrewBreyen, 2020-05-01:
@aivarannamaa Any chance this is supported?

Ivoz, 2020-05-07:
Have you exhausted all options in https://github.com/thonny/thonny/wiki/DeploymentOptions ?

It might also be possible with https://github.com/thonny/thonny/issues/755 method

aivarannamaa, 2020-05-12:
@AndrewBreyen, as Ivoz said, there may be answer in https://github.com/thonny/thonny/wiki/DeploymentOptions

Let me know again if you get stuck.

closed 2021-01-19 by aivarannamaa