thonny/thonny #1173: Matplot Doesn't work in Thonny, What is the solution??

sameerakavinthaka, 2020-04-23:
from matplotlib import pyplot as plt
plt.plot([0,1,2,3,4])
plt.show()

when i run the above codes on Thonny below errors appeared, but when i run debug it shows the graph with an error "RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility. Expected 112 from C header, got 124 from PyObject
  return f(*args, **kwds)" like this

Traceback (most recent call last):
  File "C:\Users\SAMEERA KAVINITHAKA\Documents\Documents\PythonCoding\pyditributenormal.py", line 8, in <module>
    from matplotlib import pyplot as plt
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\matplotlib\pyplot.py", line 40, in <module>
    from matplotlib.figure import Figure, figaspect
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\matplotlib\figure.py", line 20, in <module>
    from matplotlib import backends, docstring, projections
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\matplotlib\projections\__init__.py", line 1, in <module>
    from .. import axes, docstring, cbook
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\matplotlib\axes\__init__.py", line 1, in <module>
    from ._subplots import *
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\matplotlib\axes\_subplots.py", line 6, in <module>
    from matplotlib.axes._axes import Axes
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\matplotlib\axes\_axes.py", line 22, in <module>
    import matplotlib.mlab as mlab
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\matplotlib\mlab.py", line 56, in <module>
    import csv
  File "C:\Users\SAMEERA KAVINITHAKA\Documents\Documents\PythonCoding\csv.py", line 1, in <module>
    import pandas as pd
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\pandas\__init__.py", line 145, in <module>
    from pandas.io.api import (
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\pandas\io\api.py", line 8, in <module>
    from pandas.io.excel import ExcelFile, ExcelWriter, read_excel
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\pandas\io\excel\__init__.py", line 1, in <module>
    from pandas.io.excel._base import ExcelFile, ExcelWriter, read_excel
  File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\pandas\io\excel\_base.py", line 9, in <module>
    from pandas._libs.parsers import STR_NA_VALUES
  File "pandas\_libs\parsers.pyx", line 12, in init pandas._libs.parsers
ImportError: cannot import name QUOTE_MINIMAL

aivarannamaa, 2020-04-24:
It looks like you have a file `csv.py` in your working directory, which is shadowing a standard library module. Rename it to something else.

sameerakavinthaka, 2020-04-24:
Thank you aivarannamaa,
I renamed that csv.py , then worked the plot. How did you exactly said that this csv.py is the problem?
If you don't mind let me know.

aivarannamaa, 2020-04-24:
I saw it from this part of your stacktrace:

```
File "C:\Users\SAMEERA KAVINITHAKA\AppData\Roaming\Python\Python37\site-packages\matplotlib\mlab.py", line 56, in
import csv
File "C:\Users\SAMEERA KAVINITHAKA\Documents\Documents\PythonCoding\csv.py", line 1, in
import pandas as pd
```

it says matplotlib tried to import `csv` and ended up getting your script instead.

sameerakavinthaka, 2020-04-24:
Dear Aivar Annamaa,

Importing matplotlib try to import my directories or does it design to
import csv files if available inside directory &  normally we asked csvread
syntax to read csv file.

Can we do Matrix calculations using python as we do with R IDE?

Thank you for your support.

aivarannamaa, 2020-04-24:
I didn't fully understand your question, but it looks like they are not Thonny-specific, so I recommend you ask them elsewhere, eg. in Stackoverflow.

closed 2020-04-24 by aivarannamaa