thonny/thonny #1178: accessibility for screen reader for visually impaired person

imanujj, 2020-04-27:
hello sir, madam 
your Ide is not accessible for blind person. because it's not working with screen reader for visually impaired person. like NVDA, jaws and Microsoft narrator.
when i open Thonny Ide my screen reader doesn't speak anything.
please make accessible Thonny ide. if possible 
it will be a great help.
thanks.

aivarannamaa, 2020-04-27:
This is in my TODO list, but it is huge task, because the UI toolkit (Tk) doesn't support accessibility.

uharoon065, 2022-02-12:
its realy a shame  that you people treat accessibility as a after thought like developers who are blind or have any  disability are from some other planet

aivarannamaa, 2022-02-15:
Indeed, when I started Thonny as a simple visualization tool for explaining recursion, I forgot to think about potential blind users. When Thonny became more IDE-like, it was already too late to change the toolkit.

You're welcome to help and change this -- both Thonny and Tk can be developed with other IDE-s which do have support for the screen readers.