thonny/thonny #1180: Indent/Dedent doesn't work when rows selected via line number column

Ferrangie, 2020-04-29:
Thonny 3.2.7

Indent/Dedent doesn't work when lines of code are selected via the line number column.

First, I recognize that selecting lines of code via the line number column is NOT the same as selecting via the editor window, and this is probably the reason indent/dedent doesn't work.

I expected it to work.

I suggest trapping the error messages and either ignoring it or offering an explanation to the user to be certain they are selecting code via the editor, not the line number column. Or, if you're really ambitious and you have lots of time to spend, make selecting lines of code via the line number column the same as selecting from the editor!

This makes it more friendly for new users.

I can't tell you how much I love Thonny. I've been hacking the color identifiers to customize my interface. I don't know how hard/easy it would be to open up the UI to that level of customization for a slightly more advanced user.

Thanks for everything!

Ferrangie, 2020-04-29:
I forgot to include error reports:

Traceback (most recent call last):
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 969, in dispatch_from_menu
    dispatch(None)
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 911, in dispatch
    handler()
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\commenting_indenting.py", line 115, in _cmd_dedent_selection
    text.dedent_region()
AttributeError: 'Text' object has no attribute 'dedent_region'

Traceback (most recent call last):
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 969, in dispatch_from_menu
    dispatch(None)
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 911, in dispatch
    handler()
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\commenting_indenting.py", line 109, in _cmd_indent_selection
    text.indent_region()
AttributeError: 'Text' object has no attribute 'indent_region'

aivarannamaa, 2020-05-12:
Sorry for late reply. I'll look into this later.

aivarannamaa, 2020-06-04:
The problem was that the focus went to the gutter, which is itself an text widget and the command tried to indent / dedent in this widget. I changed it such that the real editor gets focused when the mouse is moved in the gutter.

TODO: need to test breakpoint placement in all platforms.

aivarannamaa, 2021-02-26:
Seems to be fixed now

closed 2021-02-26 by aivarannamaa

Ferrangie, 2021-03-07:
Thank you Aivar.
I love Thonny.