thonny/thonny #1182: Internal Error install package from local file

Alphina1, 2020-05-02:
I am trying to install a package from a local file. When I go to tools>package manager>install from local files, I get the following internal error:

Traceback (most recent call last):
  File "/usr/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/plugins/pip_gui.py", line 658, in _handle_install_file_click
    initialdir=get_workbench().get_local_cwd,
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1920, in askopenfilename
    return _get_dialog_provider().askopenfilename(**options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1960, in askopenfilename
    args = cls._convert_common_options("Open file", **options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1994, in _convert_common_options
    filename = _options_to_zenity_filename(options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2036, in _options_to_zenity_filename
    return options["initialdir"] + os.path.sep
TypeError: unsupported operand type(s) for +: 'method' and 'str'

What am I doing wrong here?

Thanks.

aivarannamaa, 2020-05-04:
Thank you for the report! You did nothing wrong -- you just stumbled upon a bug. I'll include the fix in the next release.

closed 2020-05-04 by aivarannamaa

Dushyantc1, 2021-02-02:
Hi, I am getting error while using the same thing. The error is:

Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1883, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/plugins/pip_gui.py", line 655, in _handle_install_file_click
    filename = askopenfilename(
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1920, in askopenfilename
    return _get_dialog_provider().askopenfilename(**options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1960, in askopenfilename
    args = cls._convert_common_options("Open file", **options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1994, in _convert_common_options
    filename = _options_to_zenity_filename(options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2036, in _options_to_zenity_filename
    return options["initialdir"] + os.path.sep
TypeError: unsupported operand type(s) for +: 'method' and 'str'

What I am doing wrong here. Please help.

aivarannamaa, 2021-02-02:
@Dushyantc1, please see the comments under #1583

commit 3eccbb58 (2020-05-04): FIXED #1182: pip-GUI error on Linux when installing from local file