thonny/thonny #1185: Debugging fails when I try to step-into(F7) a line with an f-string.

afcoSalazar, 2020-05-03:
Kudos for such a great work; Thonny is an excelent asset. Thanks in advance.

This is the text of the error:
Traceback (most recent call last):
  File "C:\Users\alvar_000\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1579, in event_generate
    handler(event)
  File "C:\Users\alvar_000\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 1095, in _handle_debugger_progress
    _current_debugger.handle_debugger_progress(msg)
  File "C:\Users\alvar_000\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 248, in handle_debugger_progress
    self._main_frame_visualizer.update_this_and_next_frames(msg)
  File "C:\Users\alvar_000\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 373, in update_this_and_next_frames
    self._update_this_frame(msg, frame_info)
  File "C:\Users\alvar_000\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 556, in _update_this_frame
    FrameVisualizer._update_this_frame(self, msg, frame_info)
  File "C:\Users\alvar_000\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 432, in _update_this_frame
    self._expression_box.update_expression(msg, frame_info)
  File "C:\Users\alvar_000\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 619, in update_expression
    whole_source, frame_info.filename, frame_info.current_root_expression
  File "C:\Users\alvar_000\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 717, in _load_expression
    index2 = _create_index(node.end_lineno, node.end_col_offset)
AttributeError: 'Str' object has no attribute 'end_lineno'

aivarannamaa, 2020-05-04:
Please post the code that causes it. Also your Thonny version.

afcoSalazar, 2020-05-04:
Thonny 3.2.1
Windows 8.1(64-bit)
Python 3.7.4 (32-bit)
Tk 8.6.9
###############################
nombre ="Juan"
print(f"Hola {nombre}, buen día")
pass
########
Thanks in advance

aivarannamaa, 2020-05-05:
I can't reproduce this with newest version (3.2.7). Please check whether this version works for you!

closed 2021-01-19 by aivarannamaa