thonny/thonny #1188: Thonny close my session

jucavi, 2020-05-08:
When I tried to open Tools-Options my user session is close, same happens with Run-Select interpreter for example. Logs in Thonny data folders are empty. I'm using macOS mojave. Any idea?

aivarannamaa, 2020-05-08:
Do you mean Thonny window gets closed? Which Thonny version are you using? Can you open "Help => About" ?

jucavi, 2020-05-08:
Thony version 3.2.7
It's close mac user's session

aivarannamaa, 2020-05-09:
Seems to be related to #871

jucavi, 2020-05-09:
Solved installing macOS Catalina

aivarannamaa, 2020-09-22:
@jucavi, can you please check your "Prefer tabs when opening documents" setting under you Mac's system settings (in section "Dock")? What does it say?

aivarannamaa, 2021-01-14:
Similar to #1199

aivarannamaa, 2021-02-27:
Please try whether https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg works better for you. It's with an experimental Universal 2 variant of Python 3.9.2.