thonny/thonny #1191: Micropython Objects 'leak' into Shell

T-Wilko, 2020-05-13:
Hi,

Not sure at all what's going on here but I believe it's an issue with Thonny as I've tried this with just a serial connection through PuTTY and it works fine.

I'm running micropython on an esp32 board and connecting to it through Thonny's integrated interpreter selecter. Once my main.py script runs successfully, the REPL appears to be waiting on idle for a few seconds, then it gets flooded with the following

```
<thonny>{'width': '400', 'e': '<EPD object at 3f819260>', 'buf': "bytearray(b'\\xff\\xff\\xff\\xff\\xff\\xff\\xff\\ ... 

(so on and on, as this is a framebuffer object for a 400x300 screen)

... \\xff\\xff\\xff\\xff')", 'gc': "<module 'gc'>", 'primary': '0', 'fb': '<FrameBuffer>', 'white': '0', 'x_mid': '200.0', 'bdev': '<Partition type=1, subtype=129, address=2097152, size=2097152, label=vfs, encrypted=0>', 'epaper': "<module 'epaper' from 'epaper.py'>", 'real_width': '400', 'black': '1', 'secondary': '1', 'height': '300', 'framebuf': "<module 'framebuf'>", 'y_mid': '150.0', 'uos': "<module 'uos'>"}</thonny>>
```

'epaper', 'primary', 'secondary' etc are all objects that are created in my program. I'm not sure why Thonny is 'printing' this into shell for me to see?
Like I said, PuTTY does not show this output in serial at all so it is not being outputted from my micropython device.

Any advice/hints would be great,
Thomas

T-Wilko, 2020-05-13:
My apologies, appears to be an issue with my wiring, not Thonny.

aivarannamaa, 2020-05-13:
I've seen this before. Actually I consider it to be Thonny's fault -- current version is not robust enough to handle occasional "stuttering" of serial communication with some devices. 

I've started with refactoring of the communication component, but it will take some time before it is ready to be released.

aivarannamaa, 2020-05-13:
Related to https://github.com/thonny/thonny/issues/1096#issuecomment-591488639

T-Wilko, 2020-05-14:
I think it's worth noting though, unlike #1096, my issue appears consistently in Thonny, even though the PuTTY serial only occasionally 'prints' odd data streams.

Looking forward to 3.3 update, thanks for developing my favourite IDE @aivarannamaa

T-Wilko, 2020-05-17:
To anyone else that stumbles across this issue in MicroPython and this thread, it seems to be linked to some sort of memory usage issue that can be solved by importing garbage collection and running the collection method when possible.

```
import gc

# run collection when needed, e.g. after handling large buffer object
gc.collect()
```

aivarannamaa, 2020-10-27:
Should be fixed in https://github.com/thonny/thonny/releases/tag/v3.3.0b2
Please reopen if still a problem.

closed 2020-10-27 by aivarannamaa