thonny/thonny #1192: Poor UX with 3.2.7

pulkin, 2020-05-14:
1. You moved file upload as a boot/main script into a third-level menu. Previously I was able to just use a shortcut. This is a deal breaker for me: if I would be interested in such an obfuscated navigation I would just install KDE or even Windows.

2. Once you go all through that and press "Save" button a confirmation window should pop out. Not mentioning a strong attack of nostalgia from 90s, this dialog box does not acquire focus on my system Gnome / Fedora 32. I.e. it remains behind the file dialog which is ridiculous.

pulkin, 2020-05-14:
Ah I think I got it: I am supposed to save the thing on the device and then to use Ctrl-s to upload it again and to never see those layers of dialog boxes. This partially invalidates p. 1.

aivarannamaa, 2020-05-14:
Dialogs occasionally getting hidden behind other windows is a problem of Tk. I'll see whether I can find a workaround.

BTW. Issues written with pejorative tone automatically get lower priority. I believe it's the same with any open-source project.

pulkin, 2020-05-14:
I am so sorry for that. Please do not consider it personal.

aivarannamaa, 2021-01-21:
More ideas here: https://github.com/raspberrypi/thonny-pico/issues/12

closed 2022-09-27 by pulkin