thonny/thonny #1193: Does Thonny support packages?

Jack15911, 2020-05-15:
I'm unable to create a package using Thonny (as installed on Raspberry Pi 3B+).

Thonny Tools->Manage Packages brings up a dialog that appears to be mainly for installing packages from elsewhere, but I want to create a package from a beginner-level tutorial. 

I created a directory in /home/pi/.local/python3.7/site-packages and added a new directory and to that added an empty file titled "__init__.py".

Then I attempted to save a Thonny file there, but could not. Then I created a directory using file manager and attempted to save the Thonny file "save-as," but it still didn't work.

I expect this is possible, but I'm unable to find any documentation about how to create packages.

aivarannamaa, 2020-05-15:
> Then I attempted to save a Thonny file there, but could not. 

Please be more specific. Did you get an error message?

> Then I created a directory using file manager and attempted to save the Thonny file "save-as," but it still didn't work

Same here

Jack15911, 2020-05-16:
Consider two cases, both attempting to save to /home/pi/.local/python3.7/site-packages/mypackage. (mypackage has __init__.py in it.) I want to write and save a module in mypackage.

Case 1. I write the module and then attempt to run it. System offers me a Save and I successfully save the module in mypackage.

Case 2. I write the module and attempt to save it before running it. I navigate to the mypackage but I cannot save it there - the "Okay" button is greyed out. My only choice is "Cancel." I believe I can do this in non-package actions in ~/pi, but not in package actions.

aivarannamaa, 2020-05-16:
> Case 1.

Do I understand correctly that everything is fine with this?

> Case 2.

I can't reproduce this. I opened Thonny 3.2.6 on RPi, created new file, added a definition, issued Ctrl+S, navigated to `/home/pi/.local/lib/python3.7/site-packages/mypackage` (you forgot `lib` in your example), created and entered new folder and saved file as `__init__.py`. After restarting the backend process (red Stop/Restart button), I could import my package.

Jack15911, 2020-05-16:
Case 1.  You are correct; everything is fine.

Case 2. For what it's worth, I did not use Ctrl+S to save - I used the down arrow on the GUI. The init file was present. I am unable to reproduce the error now. Close.

closed 2020-05-16 by Jack15911