thonny/thonny #1195: Editor auto-complete causes crash with MicroPython interpreter

T-Wilko, 2020-05-19:
I'm assuming this is unintentional since it results in a complete freeze of Thonny.
To give context, I have the interpreter chosen as MicroPython (esp32), but I don't have the device connected (intentionally) so it reads the following in the shell:

```
Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode)
or choose "Tools → Options → Interpreter" to select the port manually.
```

As I then proceed to use Thonny to edit some local .py files, upon attempting to auto-complete an already defined variable by pressing tab, Thonny completely freezes and remains unresponsive indefinitely. I notice that this is solved by switching to Thonny's integrated interpreter before attempting to press tab.

I'm going to go out on a limb and say that pressing tab leads Thonny to await a response from the non-existent MicroPython device's interpreter?

I'm running Thonny version 3.2.7

aivarannamaa, 2020-06-02:
Sorry for late reply!

It may be what you suspect, but in the past the Jedi library, which is used for code completions, has been acted weirdly by itself as well. I'll look into this.

aivarannamaa, 2020-10-28:
Unfortunately I can't reproduce this. Thonny 3.3 will be out soon, hopefully it fixes this for you as well.

aivarannamaa, 2020-11-26:
Thonny 3.3.0 is now out (and 3.3.1 following soon). Please reopen if you still experience this issue

closed 2020-11-26 by aivarannamaa