thonny/thonny #1196: pynput not working

fdkjgudogh, 2020-05-20:
i am trying to use pynput and micropython to send key inputs to my computer by it doesnt recognise the module pynput even after installing it on cmd

aivarannamaa, 2020-05-20:
Which OS and Thonny version are you using? Do you have a system-wide Python installation?

fdkjgudogh, 2020-05-20:
i have thonny 3.2.7 installed. what do you mean by system wide installation, sorry im new to programming

aivarannamaa, 2020-05-20:
I assume you installed the variant of Thonny which comes together with Python interpreter. In this case the expected way to install extra packages is via Tools => Manage packages.

If you have installed Thonny (with bundled interpreter) and also a separate Python interpreter, then you may have installed `pynput` to this other interpreter. 

> it doesnt recognise the module pynput even after installing it on cmd

Can you give me more details on how did you install it?

fdkjgudogh, 2020-05-20:
i cant click the on the manage packages. i downloaded pynput from the command prompt by typing pip install pynput

aivarannamaa, 2020-05-20:
> i cant click the on the manage packages. 

By change, are you using MicroPython back-end? I'm pretty sure `pynput` won't work there.

> i downloaded pynput from the command prompt by typing pip install pynput

what do you get when you execute `pip --version`?

fdkjgudogh, 2020-05-20:
pip 20.1.1 when executed.

aivarannamaa, 2020-05-20:
> pip 20.1.1 when executed.

I believe it also says the location where pip is installed. Please tell me this as well

fdkjgudogh, 2020-05-20:
c:\users\robin\appdata\local\programs\python\python38\lib\site-packages\pip (python 3.8)

aivarannamaa, 2020-05-20:
From this I infer that you have separate Python in your system besides Thonny's Python and you installed your package for this external Python. Your options:

* Make Thonny use this external Python 3.8 (Run => Select interpreter)
* Install the package for Thonny's Python (Tools => Manage packages (can't be used when MicroPython back-end is selected) or with Tools => Open system shell)

fdkjgudogh, 2020-05-20:
what do i do after opening shell

aivarannamaa, 2020-05-20:
> what do i do after opening shell

enter `pip install pynput`, just like you did, but this time Thonny's Python and pip are supposed to be in PATH (ie. `pip` refer's to Thonny's pip). When you opened system shell from Thonny, you should see the full path for Python commands in the top of the window.

fdkjgudogh, 2020-05-20:
says its already satisfied

fdkjgudogh, 2020-05-20:
and is still the same

aivarannamaa, 2020-05-20:
You mentioned MicroPython, but you didn't tell me whether you are using MicroPython back-end or not. You can check this from "Run => Select interpreter".

(In the future, if you want someone to help you in debugging a technical problem, you should be more proactive in providing information. The more information you give (screenshots, stacktraces, terminal output), the easier it is to help)

fdkjgudogh, 2020-05-20:
[image]

aivarannamaa, 2020-05-20:
So, you have selected MicroPython back-end. You can't import `pynput` in MicroPython, because you installed it to regular Python, not MicroPython. Also, you can't install it to MicroPython, because it is not meant for MicroPython.

fdkjgudogh, 2020-05-20:
Oh i didn’t know that . Is there another way I could send key inputs then?

aivarannamaa, 2020-05-20:
> Is there another way I could send key inputs then?

Again, too little information :)
Send key inputs to where? Better tell me your whole project idea from the start.

fdkjgudogh, 2020-05-20:
I wanna make a macro keypad that helps me as I am doing 3d design. For example, if I press crtl alt s, it would scale the object, I want to turn it into a macro and by pressing a key on my number pad it would do I without me pressing ctrl alt s

aivarannamaa, 2020-05-20:
But why do you need MicroPython (ESP8266)?

fdkjgudogh, 2020-05-20:
The esp8266 send key inputs to the pc

aivarannamaa, 2020-05-20:
So you want to use an ESP as a HID device? There is a related discussion here: https://forum.micropython.org/viewtopic.php?t=3640

In any way you need to create two programs -- one running on ESP which forwards button or sensor inputs to the computer, and other, running on PC, which picks up these events and controls your 3d application via `pynput`.

I don't know about HID devices. If the HID device thing is too complex, then I make the ESP print input events into its serial output and at PC-s side I would use `pyserial` package for reading in these events. 

>  im new to programming

In this case the project can be quite challenging.

closed 2021-01-19 by aivarannamaa