thonny/thonny #1197: Lags

CarterSnich, 2020-05-20:
Laggy when typing, also the auto complete dialogue box or something does not pops up immediately.

CarterSnich, 2020-05-20:
It's not laggy tho when I'm typing on the buit-in Shell.

CarterSnich, 2020-05-20:
EDIT: the lag only occurs when the codes gets many, but my code right now has only 230 lines.

aivarannamaa, 2020-05-21:
Auto complete is expected to be slow first time, but typing in a 230-line program should be fast. What is your platform?

CarterSnich, 2020-05-23:
Windows 10 Pro x64

lurch, 2020-05-23:
Try fiddling with the syntax-highlighting options to see if that improves things?

aivarannamaa, 2020-05-23:
> Try fiddling with the syntax-highlighting options to see if that improves things?

Good idea! This means "Tools => Options => Editor". Try switching off optional features. If you find the one that causes the lag, please let me know as well.

BlenderChrisDW, 2020-06-09:
I have noticed the same lag. I have found that using 'Highlight matching names' or 'Highlight local variables' causes the lag. All other options seem to work fine.

CarterSnich, 2020-06-11:
The performance improved, I disabled both 'Highlighting matching names' and 'Highlight local variables'. Thanks for the tip. But, I think both features should not be disabled, 'coz it seems like it's been a tradition for every editor/IDE to have such features enabled by default. I'll just disable it tho.

About the auto-complete dialog box, yeah it was slow for the first time.

Sorry for the late response. Thanks again for the help.