thonny/thonny #1199: Thonny freezes and logs out of Mac

kgobes, 2020-05-21:
I've just downloaded Thonny onto my Mac (running Mojave, up to date) and every time I click Run -> Select Interpreter, or Tools-> Manage Packages, my Thonny system opens a white window, freezes everything on my laptop, and logs me out. I need to use the Select Interpreter to connect to a serial port. I've tried deleting the application and reinstalling. Any suggestions?

Thanks!

aivarannamaa, 2020-05-21:
It looks similar to #871 and #1188. Unfortunately I haven't been able to reproduce this myself.

Does it happen only with "Manage Packages"? What about "Tools => Options" or "Help => About" ?

kgobes, 2020-05-21:
V 3.2 does not work either. Tools-> Options also crashes, Help-> about Thonny is fine. Any other suggestions?

aivarannamaa, 2020-05-22:
Do you have latest update of Mojave? It may be something between your OS version and Tk version (the GUI toolkit used in Thonny). 

Does something from here match with your system: https://apple.stackexchange.com/questions/153235/sudden-unexpected-logouts ?

aivarannamaa, 2020-09-22:
@kgobes, can you please check your "Prefer tabs when opening documents" setting under you Mac's system settings (in section "Dock")? What does it say?

aivarannamaa, 2021-02-27:
Please try whether https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg works better for you. It's with an experimental Universal 2 variant of Python 3.9.2.