thonny/thonny #1200: Converting tabs to spaces doesn't mark the file as modified

lurch, 2020-05-23:
If I open a python file containing tabs, Thonny asks me if I want to auto-convert them all to spaces - nice! :+1: 
However if I select Yes in the dialog that pops up, Thonny _does_ convert all the tabs to spaces, but it doesn't "notice" that the file has been modified, i.e the "Save" options remain disabled.

This means that although I can run the code, debug it etc., unless I make some other edit to the file, Thonny will **again** ask me if I want to convert-tabs-to-spaces the _next_ time I open the file. This is kinda confusing given that I already told Thonny to convert-tabs-to-spaces the _previous_ time I opened the file :wink: 

IMHO when Thonny auto-converts tabs to spaces, it ought to mark the text as modified, so that you can either manually save it, or get prompted to save it when e.g. closing Thonny.

lurch, 2020-05-23:
I did a little bit of poking about and found that I can tell _if_ the tabs-were-replaced-with-spaces by modifying `direct_insert` in `EnhancedText` [here](https://github.com/thonny/thonny/blob/master/thonny/tktextext.py#L743) to:
```python3
    def direct_insert(self, index, chars, tags=None, **kw):
        old_chars = chars
        chars = self.check_convert_tabs_to_spaces(chars)
        tabs_replaced = old_chars != chars
        super().direct_insert(index, chars, tags, **kw)
```
but I didn't track down how to tell Thonny that the file should be considered "modified".
I naively tried adding
```
        if tabs_replaced:
            self.event_generate("<<TextChange>>")
```
to the end of the `direct_insert` function, but it didn't fix the problem.

aivarannamaa, 2020-05-23:
My idea was to avoid changing the file if the user only wanted to run the program. I agree that the situation gets strange when same file is loaded several times.

What do you think of keeping current behavior with tabs and modification flags, but simply omit the confirmation dialog? I could explain it with Thonny's policy of always saving with spaces, no matter the original indentation style.

Converting the tabs to spaces before displaying code allows me to the editor code simpler, so I'd like to keep this. I could do the reverse conversion on save, but it's more code and another sentence in the documentation.

ankith26, 2020-05-24:
Even I have faced the same issue, it says it is converting all tabs to spaces, but it does not show that the file has been modified and now I cannot save the file, it does the same thing if I open it again. I believe for a beginner, all this will be too weird, it would be much better if the file showed that it got modified so that users could save the file.

lurch, 2020-05-24:
> My idea was to avoid changing the file if the user only wanted to run the program.

If the user "only" wanted to run the program, I guess they'd have selected "No" when asked if they want to convert tabs to spaces?

> simply omit the confirmation dialog?

I'm not sure - would omitting the confirmation dialog (and thus *always* replacing tabs with spaces) lead to problems like those described in #656 ? :shrug:

aivarannamaa, 2020-11-11:
Made obsolete by #599

closed 2020-11-11 by aivarannamaa