thonny/thonny #1205: Error when trying to open a file on ESP8266

troublegum, 2020-05-23:
Traceback (most recent call last):
  File "C:\Users\Noname\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1598, in event_generate
    handler(event)
  File "C:\Users\Noname\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 872, in update_dir_data
    self.cache_dirs_child_data(msg["data"])
  File "C:\Users\Noname\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 315, in cache_dirs_child_data
    for parent_path in data:
TypeError: 'NoneType' object is not iterable

aivarannamaa, 2020-06-02:
I hope to fix this under #1103

aivarannamaa, 2021-01-19:
Please try Thonny 3.3 and reopen if you're still having problems.

closed 2021-01-19 by aivarannamaa