thonny/thonny #1206: Open from Micropython Device does not work everytime

abhaylolekar, 2020-05-24:


ptranter, 2020-06-01:
Hi Thonny,

I am having a similar problem, so good to see I am not alone.  I am also using ESP32 with micropython v1.12. (esp32-idf4-20191220-v1.12.bin).  In may case I can view files on ESP device, and even copy one or two files.  If I ever try to copy, say, 6 in quick succession, I get the error described above.

I have no problem using webrepl to copy files - just not as convenient.

aivarannamaa, 2020-06-01:
Similar to #1191 
I hope to fix this under #1103

ptranter, 2020-06-02:
Thanks Aivar,

 

I meant to congratulate you on a great piece of work – ideal for MicroPython, and also useful for full Python. 

I use it in my teaching, and the students all get on well with it.

 

Regards,

 

Philip

aivarannamaa, 2020-06-02:
@ptranter, I'm glad you find Thonny useful!

pyforever, 2020-06-06:
I've the same problem with an ESP8266 NodeMCU.
Thanks for fixing it quickly and congratulations for this IDE: it's really useful and effective!
Nice regards
Enzo
> 
> While opening file from micropython device, No files are displayed. List of files are displayed in the shell window in following format.
> `<thonny>{'/': {'boot.py': {'size': 139, 'kind': 'file', 'time': 1590248588}, 'main.py': {'size': 1013, 'kind': 'file', 'time': 946684828}, 'lib': {'size': None, 'kind': 'dir', 'time': 1590248292}, 'mp_wking': {'size': None, 'kind': 'dir', 'time': 1590339520}}}</thonny>>`
> I am using ESP32 - NodeMCU with firmware file - 'esp32-idf3-20191220-v1.12.bin'
> 
> Warm regards,
> Abhay Lolekar

eighta, 2020-08-07:
Same problem here

Dilepa, 2020-09-02:
Same here.

`MicroPython v1.12-688-ga93a378e9 on 2020-08-29; ESP32 module with ESP32

Type "help()" for more information.
>>> <thonny>{'/': {'wifi.py': {'size': 812, 'kind': 'file', 'time': 5518}, 'main.py': {'size': 279, 'kind': 'file', 'time': 1043}, 'yeelight.py': {'size': 2940, 'kind': 'file', 'time': 79}, 'roundrobin.py': {'size': 476, 'kind': 'file', 'time': 65}, 'test1.py': {'size': 432, 'kind': 'file', 'time': 14}, 'test4.py': {'size': 381, 'kind': 'file', 'time': 7158}, 'test3.py': {'size': 476, 'kind': 'file', 'time': 7011}, 'boot.py': {'size': 812, 'kind': 'file', 'time': 84}, 'test5.py': {'size': 1053, 'kind': 'file', 'time': 638}, 'mqtt_test.py': {'size': 1411, 'kind': 'file', 'time': 593}, 'lib': {'size': None, 'kind': 'dir', 'time': 946684800}}}</thonny>>`

AltoRetrato, 2020-09-02:
This is a know problem (#1103) that is (probably?) already fixed in the source. @aivarannamaa said that he might publish a new beta version tonight. If you run Thonny on Linux, you can also try to [run it from source](https://github.com/thonny/thonny/wiki/Running-from-the-source).

Dilepa, 2020-09-03:
i am running from windows 10. will keep a eye for beta

aivarannamaa, 2020-09-03:
Please try the beta: https://github.com/thonny/thonny/releases/tag/v3.3.0b2

Dilepa, 2020-09-04:
I have tried the beta version. No issues at all so far. 

Great job!!!!

uncoded0123, 2020-11-26:
New version works great. Thanks for creating and updating Thonny.

aivarannamaa, 2020-11-26:
Thank you for the feedback!

closed 2020-11-26 by aivarannamaa