thonny/thonny #1207: Same program behaves very differently in thonny shell and system shell

ankith26, 2020-05-24:
I had written a multithreaded sockets application, that collects user input in one thread and does socket stuff in other. I want the application to exit when I type q in the shell. So I wrote a script like the one given below (highly simplified, original is about 500 lines of code).
```
import threading
import time

run = True

def bgthread():
    global run
    while True:
        msg = input().strip()
        if msg == "q":
            run = False
        else:
            #do other stuff
            pass

def dostuff():
    while True:
        if run:
            time.sleep(0.1)
            #do other stuff
        else:
            break

if __name__ == "__main__":
    threading.Thread(target=bgthread).start()
    dostuff()
```

The issue is, when I run the script in thonny shell - program exits when I enter q.

Running the same script in system shell does not exit the program.

Finally, I fixed issue with an easy fix - but the above program is still giving inconsistent results.

ankith26, 2020-05-24:
To add to the previous post, after the above program exits in thonny shell, when I try to do something else in thonny "files" window, following error pops up in shell
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Program Files (x86)\Thonny\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "D:\Ankith\PyChess\latest\test.py", line 9, in bgthread
    msg = input().strip()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\backend.py", line 1178, in readline
    return self._generic_read("readline", limit)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\backend.py", line 1167, in _generic_read
    raise RuntimeError("Wrong type of command when waiting for input")
RuntimeError: Wrong type of command when waiting for input
```
This clearly indicates that the program has not exited, but thonny just thinks it has exited. 

And yes, 
I am on Windows 10 running Thonny v3.2.7 that comes bundled with python interpreter.

lurch, 2020-05-24:
I think you need to add a `break` after the `run = False` line? Otherwise the `while True` inside the `bgthread` function never exits, and so the Thread you create never terminates?

Also, you probably want to change the bottom part of your script to:
```python3
if __name__ == "__main__":
    mythread = threading.Thread(target=bgthread)
    mythread.start()
    dostuff()
    mythread.join()
```
as this will ensure that your program doesn't finish until `mythread` itself has finished.

ankith26, 2020-05-25:
Yes, as mentioned above I fixed the issue with a `return` after `run = False`. But my point is, the above code still behaves very differently in thonny and system shell. So I felt I had to bring this bug to the attention of thonny developer.

lurch, 2020-05-25:
If you add the `join` code that I mentioned, then AFAICT the code then behaves the same in the system shell and in Thonny.

I _think_ that your code **is** actually running the same, and the problem is actually with Thonny going back to a `>>>` prompt even though a background thread is still running, whereas with the system python it still "sits there" while the background thread is still running? :shrug:

ankith26, 2020-05-25:
Yes that’s the issue. While the main thread exited, thonny thinks program exited so it gives `>>`, but actually the program has not exited

aivarannamaa, 2020-06-02:
Thonny only closes the background process after you run a script again or press Stop/Reset or exit Thonny. It is similary to running your script with `python3 -i my_script.py`

closed 2020-06-02 by aivarannamaa