thonny/thonny #1210: Changing editor-tab removes keyboard focus from editor

lurch, 2020-05-25:
If I have two or more files open, and I'm working on both of them, then when I click on the tab to switch which file I'm working on, the "tab button" then has keyboard focus, and I have to click back into the editor-window to give keyboard-focus back to the editor before I can carry on typing.

This is slightly annoying and it would be nice if I could switch between tabs, with the keyboard focus always being given back to the editor-part of the window.

aivarannamaa, 2020-06-02:
I guess for now I can make it work as you describe it. I need to come back to this once I start preparing Thonny for blind users.

aivarannamaa, 2024-03-29:
Fixed under #2737

closed 2024-03-29 by aivarannamaa