thonny/thonny #1215: Unusable with dark desktop theme

gimplyworxs, 2020-05-31:
Thonny is unusable if you have a dark desktop theme selected, which is a pity.
Beginning at the file-opne dialogue where you can't read white font on light grey background and so on.
Why not try to get user's settings and the set not only font to white but also background to what they have?

aivarannamaa, 2020-06-01:
Thank you for the report! Which OS are you using? Can you please post a screenshot?

Achaean, 2020-08-05:
I have the same (-similar?) problem, on a Debian oldstable based system.
I can change the UI and Syntax themes, but the floating description is unreadable. Gray (?) letters, on a yellow background.

aivarannamaa, 2020-08-08:
@gimplyworxs and @Achaean , which Thonny version are you using?

gimplyworxs, 2020-08-08:
[image]
I have thonny 2.1.16 installed on Linux Mint 19.2 MATE and using a dark theme (called vertex).

Achaean, 2020-08-08:
>  @Achaean , which Thonny version are you using?

Latest (3.2.7).

saepurusass, 2020-10-24:
Ubuntu 18.04, KDE desktop, Breeze-Dark theme. Unusable for me too.
Most programs work just fine with Dark theme. But Thonny is one of a few which do not.

Looks like Thonny is not using correct colour for background. It is always white.

Thonny is installed from Ubuntu repo. 2.1.16-3, latest available at the time of posting.
I understand that Ubuntu repo does not contain the latest and greatest version. But looks like newer versions have the same issue.

Is it possible to apply some workarounds? Maybe just a quick patch in *.py files?
Or maybe use ENV variables to override something?
I'd be happy with any workaround as long as I do not have to switch themes just because of Thonny.

**KDE, with Breeze theme (Light)**
[image]

**KDE, with Breeze-Dark theme (Dark)**
[image]

aivarannamaa, 2020-10-24:
@saepurusass, Thonny 2.1 is really old. Please try binary Python+Thonny bundle instead: 

```
bash <(wget -O - https://thonny.org/installer-for-linux)
```
 or install with pip:

```
pip3 install --user thonny
python3 -m thonny
```

BTW, Ubuntu 20.04 has latest Thonny (3.2.7) in its repo.

aivarannamaa, 2021-02-27:
Related to #1658 and #1259