thonny/thonny #1216: Suggestion, not a problem

smblackledge, 2020-06-02:
Would you please consider the following enhancements:

- Allow suggestions or enhancements to be submitted.  I don't think this is the correct place, but I haven't found a better way to submit them.

- If a user highlights a consecutive number of code lines, (s)he can then move them as a group by using the arrow keys.  Visual Studio, if I remember correctly, uses Alt+up to move the group towards the top of the file, one line for each time the arrow key is depressed.  Alt+down is used to move the group towards the end of the file, one line for each time the arrow key is depressed.

- If a user highlights a consecutive number of code lines, (s)he can comment them out (add "#" characters to the front of each line) with a trigger of your choice (menu item or hot key).  This is a toggle function, so if the highlighted lines are already commented out, using the trigger again will remove the "#" characters, but only at the front of each line.

- Allow a breakpoint to be set.  The advantage of this is that your student can save time and mistakes when (s)he has a lot of code to skip before getting to the location that needs stepping.

-  Allow debugging granularity to be changed (e.g. from faster to nicer) in the middle of a run.

- Allow a "set current line" in the debugger so that if a student needs to look at something again (perhaps after changing the debug granularity - see above), (s)he doesn't need to run the entire program over again to get to the debug point.

Thanks for a great tool,
Scott Blackledge
scott_blackledge@suddenlink.net

aivarannamaa, 2020-06-02:
Thank you for the suggestions!

> Allow suggestions or enhancements to be submitted.  I don't think this is the correct place, but I haven't found a better way to submit them.

Actually, this is the correct place for suggestions.

> If a user highlights a consecutive number of code lines, (s)he can then move them as a group by using the arrow keys.  Visual Studio, if I remember correctly, uses Alt+up to move the group towards [...]

Good idea, I'll keep this in the TODO list.

> If a user highlights a consecutive number of code lines, (s)he can comment them out (add "#" characters to the front of each line) with a trigger of your choice (menu item or hot key).  This is a [...]

Already possible with Ctrl+3. See the "Edit" menu for other similar commands.

> Allow a breakpoint to be set.  The advantage of this is that your student can save time and mistakes when (s)he has a lot of code to skip before getting to the location that needs stepping.

Already possible by double-clicking on a line number.

> Allow debugging granularity to be changed (e.g. from faster to nicer) in the middle of a run.

Hard to implement. With nicer debugger you can control the granularity by choosing between Step over and Step into. Doesn't this work for you?

> Allow a "set current line" in the debugger so that if a student needs to look at something again (perhaps after changing the debug granularity - see above), (s)he doesn't need to run the entire program over again to get to the debug point.

I imagine this can be done with a breakpoint.

BTW, which platform and which Thonny version are you using?

smblackledge, 2020-06-02:
Aivar, thanks for your very quick reply.  I’m glad that you were able to clarify #1, #3, #4, and #5 have already been implemented.  I’d like to note that It wasn’t easy to find these since there is no real documentation and my skimming wasn’t good.  #1 – the word “suggestion” doesn’t seem to be anywhere on the web page, though.  That’s why I was confused.  It should be a fairly easy correction, though. 

 

The other suggestions are related to debugging.  #5 (debug granularity) assumes that you are using “fast” debug but you need to know intermediate results inside conditionals.  For example, if I write, “while len(tail) >= 1:” I would have no idea of what the current value of “len(tail)” was, especially if tail is a string with more than 20 characters.  Unless Thonny allows you to highlight “len(tail)” or other phrases in a conditional, you have to change your code to use an intermediate variable.  Allowing a change to “nice” debug could fix this issue.

 

#6 cannot be done with a breakpoint.  For example, let’s say that we’re in the middle of a for block.  If the current line is past the start of the for and you just wanted to re-run your code from the start of the for, it is far easier to just set the current line to the start of the for.  I know that something could change if I’ve already run part of the for block.  If you need a clean run, then as you suggested, setting a break point and starting over is needed.  I guess the value of this suggestion is proportional to how painful it is getting to the lines you want to see.

 

Thanks,

Scott

aivarannamaa, 2020-06-02:
Changing from faster debugging to nicer is really hard to implement (with nicer debugging the program tree goes through heavy modification before execution). In your example I would start with nicer debugger.

I'm confused about the "current line" thing. Do you mean altering natural program flow in order to repeat a part of code manually?

lurch, 2020-06-07:
> I'm confused about the "current line" thing. Do you mean altering natural program flow in order to repeat a part of code manually?

Me too. I've tried rereading what @smblackledge wrote a few times, and I think he's effectively asking for a "step backwards in time" function in the debugger? A quick search suggests you're probably looking for https://bitbucket.org/pypy/revdb/src/default/ (which I've never used myself, and unfortunately it's [not seeing much maintenance](https://bitbucket.org/pypy/revdb/commits/) ).

smblackledge, 2020-06-08:
Exactly.  Imagine:

               # do stuff to initialize str

               …

               For i in str:

                  Result = myFunc(“do this”, str, 43)

                  Print(Result)

 

I would set a break point at the myFunc() call but discover that my str initialization was wrong.  I would then, if other variables weren’t barfed on too badly, set the current line to where the str init code was.  I understand that Microsoft has a lot of resources that you don’t have, but the Visual Studio debugger has this feature.

 

Scott

aivarannamaa, 2020-06-08:
Thonny's nicer debugger does allow stepping back in time -- eg. if you made a step over and you realize step into would have been more appropriate, you can do "Run => Step back" and try again. Note that when trying again the code is not executed again -- just the intermediate states collected when you did "Step over" can be replayed with different granularity. More details here: https://comserv.cs.ut.ee/home/files/leemet_informaatika_2018.pdf?study=ATILoputoo&reference=57D36B3D851E10E6FC750A19E3479D643D8A0FB7

aivarannamaa, 2020-06-08:
>  I understand that Microsoft has a lot of resources that you don’t have, but the Visual Studio debugger has this feature.

Take this, Microsoft! :) Seriosly, Thonny's implemention is pretty naive and this is one of the reason the nicer debugger is slow and memory hungry.

smblackledge, 2020-06-18:
Btw, to answer your question from June 6, I'm using Thonny 3.2.6 and I'm on a Raspberry Pi 4B, 4 GiB (completely updated/upgraded).

aivarannamaa, 2021-02-26:
I made a new task for moving lines up and down: #1656

closed 2021-02-26 by aivarannamaa