thonny/thonny #1218: Suggestion

smblackledge, 2020-06-02:
I know that Thonny is a "student" debugger.  I also know that I'm suggesting things that are difficult to implement.  These items, though, are things I use all the time in other debuggers.  

- Documentation is in multiple places and is incomplete.  A comprehensive user's guide needs to be created. (Highest priority)

- It would be useful to allow variables to be changed during the run (for testing or "what if" purposes).

- It would be useful to allow highlighting a variable, function call, etc.  and display the value or returned value.

aivarannamaa, 2020-06-02:
> Documentation is in multiple places and is incomplete. A comprehensive user's guide needs to be created. (Highest priority)

I agree that the situation with documentation and feature discoverability is not good. I created an issue for enhancing user documentation (#1219). BTW, you are welcome to start working on this! 

About multiple places -- did you mean help pages and wiki? I looks fine to me when more technical issues are kept separate from user documentation.

> It would be useful to allow variables to be changed during the run (for testing or "what if" purposes).

It would be useful when the user already knows Python, but I believe it is better for the beginners if program construction and execution are kept separate.

> It would be useful to allow highlighting a variable, function call, etc. and display the value or returned value.

Again, this could be convenient when you already know the principles, but for the sake of beginners I don't want to mix up the static face of the program (expressions) and the runtime face of the program (values in variables table or operation stack). The ones who already understand the difference should move on to PyCharm, which has this feature.

> Thonny debugger won't display a list with more than 20 entries.

It simply truncates the string returned by value's `repr`. You can see more entries when you open View => Object Inspector and select the value in variables table (or Ctrl+click on the value in the expression box). I know, it is not documented properly :)

In conclusion -- all these features are nice to have, but I have limited time to spend on Thonny and my priority is to keep the focus on beginners.

smblackledge, 2020-06-04:
No problem.  After all, these are suggestions.  I’ll give serious thought about a proper user’s guide.

 

Thanks for your great product,

Scott

closed 2020-06-18 by smblackledge