thonny/thonny #1220: Crashes on start; reinstalling has no effect

jgordon510, 2020-06-03:
I have a student with a copy of Thonny that crashes on start.  I have no physical access to the computer, but we've tried uninstalling and reinstalling Thonny, but the issue remains.  Here is the error:
Traceback (most recent call last):
```
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 544, in _start_runner
    self._runner.start()
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 102, in start
    self.restart_backend(False, True)
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 557, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1232, in __init__
    super().__init__(clean, get_interpreter_for_subprocess())
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1062, in __init__
    super().__init__(clean, executable)
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 771, in __init__
    self._start_background_process(clean=clean)
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 836, in _start_background_process
    **extra_params
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Copperdragonfly\AppData\Local\Programs\Thonny\lib\subprocess.py", line 1207, in _execute_child
    startupinfo)
NotADirectoryError: [WinError 267] The directory name is invalid
```

aivarannamaa, 2020-06-04:
Did this error appear on first run of Thonny or the user was able to use it initially?

Are there any non-letter characters in their username or in the path of the directory where they last run a script from?

If Thonny initially worked, then please send me their Thonny's configuratin file. If Thonny's window stays up after closing the error, then open the directory with "Tools => Open Thonny data directory", otherwise it's `C:\Users\<username>\AppData\Roaming\Thonny` (AppData is hidden directory).

jgordon510, 2020-06-04:
He was able to use it initially.  It happened after he reorganized his
projects, changing some file names and folders.

I'll double check on the non-letter characters.  I'll send the
configuration file next week when I work with him again.

jgordon510, 2020-07-14:
Sorry for the late update.  I ended up just deleting the data directory.  It appeared to have been caused by a reference to a moved folder in the recent files.

I'd be happy to investigate further if this is something you'd consider worthy of fixing.

aivarannamaa, 2020-07-14:
I'll go through configuration options and see whether I notice something that must be guarded against mistakes. Thanks for your report!