thonny/thonny #1222: Investigate setting breakpoints inside threads

aivarannamaa, 2020-06-04:
https://docs.python.org/3/library/threading.html#threading.settrace

aivarannamaa, 2020-06-04:
Currently Thonny only supports stopping the main thread. It should be possible to extend the support to all threads, but I'm afraid it would require serious refactoring and I'm not sure how useful it would be in Thonny (the UI would need to be able to show multiple stacks etc.)

An easier approach would be allowing stepping only inside a single thread (be it main or secondary), eg. the one that gets to a breakpoint first. This would require less changes in the UI and back-end.

lurch, 2020-06-07:
Multi-threaded code is always a tricky problem - perhaps it should be considered "beyond" Thonny's focus on beginners? (as you mentioned in a recent issue)

aivarannamaa, 2020-06-08:
@lurch , I agree that it's tricky. I just wanted to write my thoughts down after someone asked me about this possibility.

From the user experience viewpoint, supporting stopping inside a single secondary thread shouldn't make the UI more complex. Therefore I'll keep this task open in case there appears a library or framework which is desired by the beginners and which does its work in a thread for some reason. Or, in case I feel irresistible urge to solve this problem :) (which I currently don't).

lurch, 2020-06-08:
I guess writing a GUI (e.g. in tkinter) might be the place that most beginners first come across threads? :man_shrugging:

aivarannamaa, 2020-06-08:
Basic usage of Tkinter doesn't involve secondary threads. Tkinter can reliably run only in the main thread. You only need to create a thread for some long process (eg. file download).

nicolasdespres, 2020-07-24:
I am working on a framework called [Youpy](https://github.com/nicolasdespres/youpy) that uses thread to run user code. I plan to use Thonny with this framework for teaching Python in my class. Since users write their code in a callback triggered on an event, all the user code is executed in a different thread (one per event).

For me it is critical to be able to have Thonny able to stop on breakpoints set in user thread. Without that, my student will have to fallback to the good old debugging technique of the log file, which is sad considering Thonny's advanced debugging features.

Maybe it would be enough at the beginning for the stack view to show only the thread currently being debugged.

aivarannamaa, 2020-07-25:
@nicolasdespres, Youpy looks really interesting! I've been thinking about how I would make "Scratch in Python", but I'm afraid I'll never reach actions with this.

I'll make some experiments -- maybe this issue is not so difficult than I have thought.

nicolasdespres, 2020-07-25:
Thank you for your feedback and considering doing some experiment :)