thonny/thonny #1224: Problem when handling 'get_dirs_child_data_response'

engmarsa88, 2020-06-05:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 1598, in event_generate
    handler(event)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\base_file_browser.py", line 872, in update_dir_data
    self.cache_dirs_child_data(msg["data"])
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\base_file_browser.py", line 315, in cache_dirs_child_data
    for parent_path in data:
TypeError: 'NoneType' object is not iterable

engmarsa88, 2020-06-05:
Platform is ESP32 and 'main.py' is running properly but I can not see other files in ESP32

aivarannamaa, 2020-06-06:
I hope to solve this under #1103

engmarsa88, 2020-06-06:
Thanks for the response. Would you please advise me how this solved? Should I just copy the py files into where? I read #1103 but I couldn't find way to solve it

Thanks a lot!

aivarannamaa, 2020-06-06:
@engmarsa88, I meant that I need to solve this bug.

aivarannamaa, 2021-01-19:
Please try Thonny 3.3 and reopen if you're still having problems.

closed 2021-01-19 by aivarannamaa