thonny/thonny #1225: Strange random behaviour of Thonny

pyforever, 2020-06-06:
Hi,
I found sometimes, after making a change to code and saved it to both pc and ESP8266, running the code executes the previous saved version of it (without the last changes).
The strangest thing (to me) is that if I reload the code both from pc and board they HAVE the changes I made.
The problem persists after some resets and sometimes restarting Thonny IDE. Then suddenly, after one more board reset,  the correct code is executed and then at every board reset it never fails anymore.

aivarannamaa, 2020-06-06:
Are you saving your code to main.py or another module which is imported from main.py?

pyforever, 2020-06-07:
> 
> 
> Are you saving your code to main.py or another module which is imported from main.py?

I see this behaviour in both cases: changing code in main or/and in my imported module.

aivarannamaa, 2020-06-08:
Currently I don't have any ideas of what may cause this. Please let me know if you find a way to reliably reproduce this!

pyforever, 2020-06-08:
Ok, I'll try to write a small reproducible code so you can see this.

aivarannamaa, 2021-01-19:
Please try Thonny 3.3 and reopen if you still see this behavior

closed 2021-01-19 by aivarannamaa