thonny/thonny #1226: Packages install in rel 3.2.7

pyforever, 2020-06-06:
Hi,
I need to install PySerial on my ESP8266 NodeMCU but Tools menu "Manage packages" is dimmed: How can I install it?
May I download and copy it to Thonny other libraries folder? If so: where is it located?
Thanks.

aivarannamaa, 2020-06-06:
I don't think PySerial works on ESP8266. As far as I know it only works on Windows, Linux and macOS.

pyforever, 2020-06-06:
Thanks, no package to manage serial communication with ESP8266? I need to send receive small files (a few hundred bytes at most). However: How can I install new libraries in Thonny?
Nice regards.

pyforever, 2020-06-06:
> 
Thanks, no package to manage serial communication with ESP8266? I need to send receive small files (a few hundred bytes at most). However: How can I install new libraries in Thonny?
Nice regards. 

> I don't think PySerial works on ESP8266. As far as I know it only works on Windows, Linux and macOS.

aivarannamaa, 2020-06-07:
If you wan't to communicate with your ESP from your PC, then you need to have PySerial installed to the Python interpreter on your PC. If you are using the same interpreter which is running Thonny (Run => select interpreter => same interpreter which runs Thonny), then you already have PySerial available, because Thonny relies on it.

Once you select this interpreter, "Tools => manage packages" becomes available.

pyforever, 2020-06-07:
I need to establish a communication between ESP8266 and another board YARM based (my project).
YARM has both serial and USB: it would be fine for me using USB to exchange small files. Communication will asynchronous: YARM  and 8266 will send data each others in unpredictable times.
Both boards will be usually busy with other chores so data received from both will be processed in one second more or less.

lurch, 2020-06-07:
> I need to establish a communication between ESP8266 and another board YARM based

That sounds like something outside the scope of the Thonny editor?

closed 2020-06-08 by aivarannamaa