thonny/thonny #1228: More issues

smblackledge, 2020-06-08:
Features I really do use in other debuggers (e.g. Visual Studio).  It would be nice if the debugger supported:

- watchpoints

- triggering a breakpoint/watchpoint after the nth time.  If code fails the 307th time through a loop, skip the first 306.

- displaying numbers in other bases.  For example, hex and binary

- displaying line numbers in the debug window

- setting another breakpoint when at an existing breakpoint

- ERROR - clicking above or below the movable thingy in a slider widget doesn't do anything

- I'm reporting this without trying it again, but I remember using "sys.exit(main())" at the end of a program.  This causes a traceback to be displayed (something wasn't handled correctly) when single-stepping but it executes correctly at full speed.

- ctrl-? would cause a keyboard interrupt which would enter something like a breakpoint.  This would allow variables to be displayed/examined, etc.  You would have to have started running with the debugger.  I'm not suggesting hitting the green arrow to start running and then expecting to trigger a breakpoint.  I'm guessing that you'd have to be running with the "nice" or "fast" debugger selected.

- I'm running a practice problem from Topcoder (ABBAReplace problem).  This generates strings that are 7M in length.  Thonny will hang.  I don't think it is running out of memory.  I don't have any idea why the hang, though (ran out of some resource?).  Here's the source (what I was debugging when I encountered most of the items in this email).  Please name it ABBAReplace.py.  It will run a *long* time.

Scott

################################################################################
# Consider a very simple string rewriting system. We have a string over the
# alphabet {A,B}. In each step, we find all occurrences of "AB" in the current
# string and we change each of them into "BA".
# 
# For example, if we start with the string "AABABB", the rewriting will go
# as follows:
# 
# 0: AABABB
# 1: ABABAB
# 2: BABABA
# 3: BBABAA
# 4: BBBAAA
#
# At this point, the rewriting has stopped, as there are no more "AB"
# substrings in the current string.
# 
# Use the following pseudocode to generate a string S of length N:
# 
# state = seed
# S = Sprefix
# while length(S) < N:
#     state = (state * 1103515245 + 12345) modulo 2^31
#     if state < threshold:
#         S += 'A'
#     else:
#         S += 'B'
#
# Given the string S, compute and return the number of rewriting steps.

testcases = [
#        Sprefix, N, seed, threshold, expected
        ["AABABB", 6, 0, 0, 4],
        ["", 0, 4, 7, 0],
        ["ABBABAABABBBABBBB", 17, 0, 0, 11],
        ["AABAA", 17, 47474747, 1000000000, 10],
        ["", 7000000, 47, 1123456789, 3661759],
        ["", 7000000, 47, 1987654321, 6479030],
        ["", 7000000, 47, 471, 5849431],
        ["B", 1, 1847635854, 2147233648, 0],
        ["B", 2, 1797583853, 2196388, 0],
        ["ABB", 3, 1355395163, 486381, 2],
        ["ABBA", 4, 1607036031, 2140324368, 2],
        ["AAAAB", 5, 1558380042, 2147020398, 4],
        ["AABAB", 6, 1516333974, 3298058, 4],
        ["BBABBBB", 7, 489349967, 2146000780, 4],
        ["BBABA", 8, 2046595582, 2084695727, 1],
        ["ABBBAAABB", 9, 758823778, 45788, 5],
[... 144 lines cut ...]
                loc = S.find("AB", i)      # Anything left to do?
                if loc < 0:
                    break
            
        return count    

def main():
    cases = 0
    cl = ABBAReplace()
    for Sprefix, n, seed, threshold, expected in testcases:
        
        cases += 1
        print(f"cases = {cases}, expected = {expected}")
        
        result = cl.countSteps(Sprefix, n, seed, threshold)
        if expected != result:
            print("\nError")
            print(f"Expected {expected}, received {result}")
            break
    print("Done")
    return result

    
if _name_ == '_main_':
    main()

smblackledge, 2020-06-08:
I'm not confident that garbage collection is being done (or being done correctly).  That may be related to why Thonny hangs.  The strings that the program creates are huge and it creates a lot of them.  (Yes, the actual solution does not actually change occurances of AB to BA.  I've got another solution that works.

Sorry for the weird formatting in the email.  It is late and I'm tired.  I can put the program in a file and attach it if you want.

Scott

smblackledge, 2020-06-08:
I was going to try to fix the formatting by editing the post.  That window showed the code correctly.  Oh well, please tell me if there are problems with the code and I'll quickly get a correction to you.

aivarannamaa, 2020-06-08:
> watchpoints

I haven't used watchpoints myself. It looks like a feature for advanced users, but I still want to understand it better. As far as I understand, this could be done such that the user provides an expression and the debugger tries to evaluate it on **each** possible stopping point. Is the expression always evaluated against the global environment of the main module? Or against whatever environment the execution is in currently? When the execution moves from global scope to local, the meaning of the expression can change and therefore its value can change as well -- should the debugger stop in this point? When the execution moves to another scope and the expression is not valid anymore, does this count as change in the value?

> triggering a breakpoint/watchpoint after the nth time. If code fails the 307th time through a loop, skip the first 306.

Maybe allowing general conditions on breakpoints would be more useful? In either way, it would be advanced feature and I'm not sure Thonny's target group needs this.

> displaying numbers in other bases. For example, hex and binary

Good idea and easy to implement! I created separate issue for this: #1230

> displaying line numbers in the debug window

Do you mean line numbers in the function call windows? This should be done indeed (https://github.com/thonny/thonny/issues/1231)

> setting another breakpoint when at an existing breakpoint

This should work. Have you experienced problems with this?

> ERROR - clicking above or below the movable thingy in a slider widget doesn't do anything

Don't understand what are you talking about.

> I'm reporting this without trying it again, but I remember using "sys.exit(main())" at the end of a program. This causes a traceback to be displayed (something wasn't handled correctly) when single-stepping but it executes correctly at full speed.

Please create separate issue with example code when you can reproduce this.

> ctrl-? would cause a keyboard interrupt which would enter something like a breakpoint. This would allow variables to be displayed/examined, etc. You would have to have started running with the [...]

I suppose we could call it "Pause". This has been suggested before, but now it occurred to me that it might be useful for trying to understand why the program ended up in an infinite loop. I created a separate issue: https://github.com/thonny/thonny/issues/1232

> I'm running a practice problem from Topcoder (ABBAReplace problem). This generates strings that are 7M in length. Thonny will hang. I don't think it is running out of memory. I don't have any idea [...]

Thonny does hinder garbage collection, because it keeps references to certain objects to be able to display the Heap table. I'll see what can be done (https://github.com/thonny/thonny/issues/1233)

smblackledge, 2020-06-08:
I’m going to clone the source which will help when writing a user guide.

 

I’ve got to start numbering these requests when I initially ask for something :).

 

I’ll start by suggesting a quick way to talk about what watchpoint environments; until we have something better, I’ll use a letter: the global environment (g), the current thread (t), whatever is using the current stack frame (f), or maybe even the current block of code (b).

 

#1) Would the watchpoint be considered global or local?  If a global resource is being watched, then the expression is evaluated at every stop.  If a local resource, only evaluate the expression when the correct environment is in effect.  If you can’t determine locality, treat the resource as global.  Scope change does not mean the watch has been triggered.  Is this answer good enough?  If no, I can be more thorough.

 

#2) Please reconsider (only break on the nth occurrence).  This won’t be used often, but it will be used.  For breakpoints, I don’t consider it an advanced feature.  This is very useful in a loop where you may or may not know which iteration has the problem.  It is probably an advanced feature when used with watchpoints, though.

 

#4) Yes.

 

#5) No, it doesn’t work.  I set a breakpoint and ran the debugger.  The debugger then stopped at the breakpoint.  Double-clicking on another code line in the code window doesn’t do anything, especially it does not set a new breakpoint.  I don’t know if this problem has to do with the line numbers not being displayed and therefore not having a prefix area to click in to (re)set a breakpoint.

 

#6) Sorry.  I was too tired to write that one and I did too poor a job of explaining.  Hopefully this attempt is better.

 

The slider widget on the right side of the code editor window has the up- and down-arrows and the “bar”, which you can left-click and drag to view different places in the source code.  The arrows and the bar both work, but clicking above the bar and below the bar, thinking I can get about a page’s movement, doesn’t do anything. 

 

#7) I wrote a small test program but cannot get it to fail.  I’ll ignore this for now.

 

#8) That’s exactly the reason I wanted the keyboard to trigger a breakpoint.

 

#00) I’m reviewing Step Back in the document tonight.

 

Scott

aivarannamaa, 2020-06-10:
> I’ve got to start numbering these requests when I initially ask for something :).

It's better if you create separate issue for each topic. I now created some new issues so please write your comments under these.

> If a global resource is being watched, then the expression is evaluated at every stop.  If a local resource, only evaluate the expression when the correct environment is in effect.

I created a new issue for discussing watches: https://github.com/thonny/thonny/issues/1234

> Please reconsider (only break on the nth occurrence)

Here's a new issue regarding breakpoint properties: https://github.com/thonny/thonny/issues/1235 . Please provide an example which demonstrates the utility of the feature you proposed.

> Double-clicking on another code line in the code window doesn’t do anything, especially it does not set a new breakpoint.

Double-clicking on a code line wouldn't set a breakpoint even before starting the debugger, you need to set it on the line number margin. For local context this depends on #1231 . As a workaround you can turn off "Tools => Options => Run & Debug => Show function calls in separate window".

> The slider widget on the right side of the code editor window has the up- and down-arrows and the “bar”, which you can left-click and drag to view different places in the source code.

I see you meant the scrollbar. Its behavior is defined by the GUI toolkit (Tk). I now confirmed that what you tried really doesn't work on Linux but it does work in Windows. Hopefully next Tk version will fix this for Linux as well

closed 2021-01-19 by aivarannamaa

kilitary, 2025-09-22:
is result hat is to ruin the cheaters?