thonny/thonny #1229: Manually installed python disables debug feature

globalturist, 2020-06-08:
I'm currently trying to manually install Python 3.8.3 with thonny 3.2.7 on windows 10 and the ide works however the debug features are grayed out and unusable.
Tried installing previous python versions (down to 3.7.6 which should be included with the thonny installer) both x64 and x32 and the same problem remains. Using pip and pip3 to install thonny doesn't solve the problem either.

NOTE: When using the thonny installer it works perfectly and debug is usable however pip is not usable and python is not shown in installed programs.

The rational for using manual install of python and thonny is to enable the ability to automate the install process and make a single installer with AutoIt that does the following in the background:
1)Install cp210x driver 
2)install python with custom options
3)installs esptool with pip
4)installs thonny

The custom installer is already written and works perfectly however debug features in thonny remain grayed out.

aivarannamaa, 2020-06-08:
As you mentioned `cp210x` and `esptool`, I assume you are dealing with MicroPython. Do you have MicroPython selected as interpreter (Run => Select interpreter) ?

Debugger is not available for MicroPython back-end, be it with manually installed Python or with bundled Python.

globalturist, 2020-06-10:
You were right ! Thanks pointing it out. After further reading I understood the debug feature is already requested for micropython therefore the issue is closed as its not an issue.

closed 2020-06-10 by globalturist