thonny/thonny #1234: Add "Watches" view

aivarannamaa, 2020-06-10:
it would allow quickly check the value of an expression each time the debugger stops and/or make the debugger stop when this value changes

It would consist of a table with three columns:

* expression (editable by the user)
* value (update by the debugger)
* checkbox "stop on change"

aivarannamaa, 2020-06-10:
I propose that the user doesn't need to specify for which context the watch expression is meant for. This means that an expression meant for global context may change meaning in a local context where a local variable shadows a global one and therefore may trigger the watch unexpectedly.

I believe this is small price to pay for not  having to worry about contexts when specifying the watches. Besides, there is a simple workaround of choosing different name for the local variable and usually when local and global variable have same name, they refer to same object anyway.

smblackledge, 2020-06-18:
Part of the value I've gotten out of watch points is that a break is triggered as soon as the value is changed or as soon as an expression is evaluated to be true.  This would mean that Thonny would need to check the value of the variable/expression at the finest granularity allowed (statement, expression, etc).  I know this will kill performance, but if the programmer needs to stop at the watch point, (s)he must be willing to give up that performance.  This performance trade-off is not unique to Thonny.

smblackledge, 2022-01-21:
Thanks for linking these two issues for me.  I'm sorry that It didn't occur to me to append to this one.

As you have seen, the two are slightly different, though.  This issue describes watching as something that triggers a break under a specified condition.  #2132 approaches the problem from an "already in single-step" direction. It may be helpful if I were to rethink how to write the request to unify the two.