thonny/thonny #1236: Autocomplete error in Thonny 3.2.7

antonvier, 2020-06-11:
Hello.
every time i try to use <tab> to autocomplete i get the following error:

Traceback (most recent call last):
   File "/usr/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 879, in _cmd_editor_autocomplete
     script = jedi.Script (cmd.source, cmd.row, cmd.column, sys_path = [self._api_stubs_path])
   File "/usr/lib/python3.7/site-packages/jedi/api/__init__.py", line 133, in __init__
     project._sys_path = sys_path
AttributeError: 'NoneType' object has no attribute '_sys_path'

but if I do the same at thonny's prompt it works.
reinstalled using pip, but kept giving "Autocomplete error"

thanks.

aivarannamaa, 2020-06-11:
Thank you for the report! I'll look into it later.

antonvier, 2020-07-11:
hi. any news?

aivarannamaa, 2020-07-12:
Haven't had chance to work on it yet.

aivarannamaa, 2020-10-28:
It seems to be a problem with a specific version of `jedi`: https://github.com/davidhalter/jedi/issues/1555

It should be fixed in a newer release.

closed 2020-10-28 by aivarannamaa