thonny/thonny #1237: Can't choose Micropython interpreter with Thonny 3.2.7

G-Boronad, 2020-06-11:
Hye, 
I have only three possibility in the options, there is no  ESP32, ESP8266 or microbit Interpreter.
[image]

Thanks

aivarannamaa, 2020-06-12:
Please select "Tools => Open Thonny data folder" (or whatever it is in French), close Thonny and check whether there is anything in frontend.log or frontend_faults.log

aivarannamaa, 2020-06-12:
Also, make sure you really have Thonny 3.2.7 (Help => About)

G-Boronad, 2020-06-12:
Hello, 
Here my version:
[image]
My log files are empty:
[image]
Thanks

aivarannamaa, 2020-06-15:
Can you try following:

* close Thonny
* Open system command prompt (open start menu, type `cmd` and press enter)
* enter following command `cd %LOCALAPPDATA%\Programs\Thonny` (if you installed for yourself) or `cd "C:\Program Files (x86)\Thonny"` (if you installed for all users)
* enter `python -m thonny`
* Select "Run => Select interpreter"
* Check the command propmt window. If there are some error messages then please paste them under this issue.

G-Boronad, 2020-06-15:
We can see these  error messages when i run thonny in prompt:

Microsoft Windows [version 10.0.18363.900]
(c) 2019 Microsoft Corporation. Tous droits réservés.

C:\Users\hp>cd "C:\Program Files (x86)\Thonny"

C:\Program Files (x86)\Thonny>python -m thonny
ERROR:root:Failed loading plugin 'thonny.plugins.circuitpython'
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 394, in _load_plugins_from_path
    m = importlib.import_module(module_name)
  File "C:\Program Files (x86)\Thonny\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\circuitpython\__init__.py", line 5, in <module>
    from thonny.plugins.micropython import (
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 24, in <module>
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'
ERROR: Failed loading plugin 'thonny.plugins.circuitpython'
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 394, in _load_plugins_from_path
    m = importlib.import_module(module_name)
  File "C:\Program Files (x86)\Thonny\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\circuitpython\__init__.py", line 5, in <module>
    from thonny.plugins.micropython import (
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 24, in <module>
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'
ERROR:root:Failed loading plugin 'thonny.plugins.esp'
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 394, in _load_plugins_from_path
    m = importlib.import_module(module_name)
  File "C:\Program Files (x86)\Thonny\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\esp\__init__.py", line 1, in <module>
    from thonny.plugins.micropython import (
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 24, in <module>
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'
ERROR: Failed loading plugin 'thonny.plugins.esp'
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 394, in _load_plugins_from_path
    m = importlib.import_module(module_name)
  File "C:\Program Files (x86)\Thonny\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
[... 47 lines cut ...]
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 24, in <module>
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'
ERROR: Failed loading plugin 'thonny.plugins.micropython'
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 394, in _load_plugins_from_path
    m = importlib.import_module(module_name)
[... cut ...]
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 24, in <module>
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'

thanks for your help

aivarannamaa, 2020-06-16:
That's weird -- it looks like for some reason Thonny can't find a library which comes with Thonny's installer.

Do you happen to have a script named `serial.py` in your working directory? If yes, then try renaming it and restarting Thonny.

G-Boronad, 2020-06-18:
Here is my directoty named "serial:
[image]
I try to renamed it but it doesn't change anything.
I don't have any "serial.py" script.

aivarannamaa, 2020-06-18:
I assume this directory is under Thonny's installation directory.

Please also check which packages you have in your user package directory. In order to open this, select "Tools => Manage plug-ins" and click the link under "Target ___" (in the lower part of the dialog). Do you have something named "serial" there?

G-Boronad, 2020-06-18:
I see a stuf,
My installation directory  with serial directory is :
C:\Users\hp\AppData\Local\Programs\Thonny\
My python default interpreter is:
[image]
But, my plugins folder is C:\Users\hp\AppData\Roaming\Python\Python37\site-packages. And the  serial plugin is installed  in this directory.
[image]

aivarannamaa, 2020-06-18:
I guess we found the reason. This package shadows the package with the same name, which is required by Thonny's MicroPython support. They have different PyPI names (`serial` vs `pyserial`), but same module names (`serial`).

If need this package under user packages, the I don't have a solution. If you can use it in a virtual environment, then uninstall it from user packages and restart Thonny.

If you need this package

G-Boronad, 2020-06-18:
I uninstalled serial plugins (i used this plugin to communicate between an arduino and a python program) and it works. Now I can see  micropython interpreters.
Perhaps il will be possible to change module name for micropython interpreter in a next version of Thonny
Thanks for your help.
G.B.

aivarannamaa, 2020-06-18:
> i used this plugin to communicate between an arduino and a python program

I'm afraid you have confused something. [According to PyPI](https://pypi.org/project/serial/), this package is "A framework for serializing/deserializing JSON/YAML/XML into python class instances and vice versa"

PySerial is for serial communication and I can't change its name.

dsl400, 2020-11-30:
is this the same problem ?

`Traceback (most recent call last):
  File "C:\Users\dsl40\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\dsl40\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\backend_config_page.py", line 122, in _backend_changed
    page = self._get_conf_page(backend_desc)
  File "C:\Users\dsl40\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\backend_config_page.py", line 141, in _get_conf_page
    self._conf_pages[backend_desc] = cp_constructor(self.labelframe)
  File "C:\Users\dsl40\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 409, in __init__
    for p in list_serial_ports()
  File "C:\Users\dsl40\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 873, in list_serial_ports
    return list(serial.tools.list_ports.comports())
  File "C:\Users\dsl40\AppData\Local\Programs\Thonny\lib\site-packages\serial\tools\list_ports_windows.py", line 412, in comports
    return list(iterate_comports())
  File "C:\Users\dsl40\AppData\Local\Programs\Thonny\lib\site-packages\serial\tools\list_ports_windows.py", line 334, in iterate_comports
    info.serial_number = get_parent_serial_number(devinfo.DevInst, info.vid, info.pid)
  File "C:\Users\dsl40\AppData\Local\Programs\Thonny\lib\site-packages\serial\tools\list_ports_windows.py", line 197, in get_parent_serial_number
    vid = int(m.group(1), 16)
AttributeError: 'NoneType' object has no attribute 'group'

`

aivarannamaa, 2020-11-30:
@dsl400, no, your problem is reported under #1443 and I hope to release the new version with the fix within this week.

closed 2021-01-19 by aivarannamaa