thonny/thonny #1238: Failed to open SPI device.

rofi2809, 2020-06-11:
in my python script, which works good in console (with command "python" file.py), the access to SPI doesnt work in ThonnyIDE (Debug).

=> %Run tes_OHW.py
Failed to open SPI device.
: Is a directory
Failed to open SPI device

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2020-06-12:
Can you post the program?

rofi2809, 2020-06-12:
Sure - it's the sample program from my LED Matrix:

#!/usr/bin/python
# -*- coding:utf-8 -*-
from ctypes import *
import time
import RPi.GPIO as GPIO

hspi = CDLL('./dev_hardware_SPI.so')

disp = [
    [0x3C,0x42,0x42,0x42,0x42,0x42,0x42,0x3C],#0
    [0x10,0x30,0x50,0x10,0x10,0x10,0x10,0x7C],#1
    [0x3E,0x02,0x02,0x3E,0x20,0x20,0x3E,0x00],#2
    [0x00,0x7C,0x04,0x04,0x7C,0x04,0x04,0x7C],#3
    [0x08,0x18,0x28,0x48,0xFE,0x08,0x08,0x08],#4
    [0x3C,0x20,0x20,0x3C,0x04,0x04,0x3C,0x00],#5
    [0x3C,0x20,0x20,0x3C,0x24,0x24,0x3C,0x00],#6
    [0x3E,0x22,0x04,0x08,0x08,0x08,0x08,0x08],#7
    [0x00,0x3E,0x22,0x22,0x3E,0x22,0x22,0x3E],#8
    [0x3E,0x22,0x22,0x3E,0x02,0x02,0x02,0x3E],#9
    [0x08,0x14,0x22,0x3E,0x22,0x22,0x22,0x22],#A
    [0x3C,0x22,0x22,0x3E,0x22,0x22,0x3C,0x00],#B
    [0x3C,0x40,0x40,0x40,0x40,0x40,0x3C,0x00],#C
    [0x7C,0x42,0x42,0x42,0x42,0x42,0x7C,0x00],#D
    [0x7C,0x40,0x40,0x7C,0x40,0x40,0x40,0x7C],#E
    [0x7C,0x40,0x40,0x7C,0x40,0x40,0x40,0x40],#F
    [0x3C,0x40,0x40,0x40,0x40,0x44,0x44,0x3C],#G
    [0x44,0x44,0x44,0x7C,0x44,0x44,0x44,0x44],#H
    [0x7C,0x10,0x10,0x10,0x10,0x10,0x10,0x7C],#I
    [0x3C,0x08,0x08,0x08,0x08,0x08,0x48,0x30],#J
    [0x00,0x24,0x28,0x30,0x20,0x30,0x28,0x24],#K
    [0x40,0x40,0x40,0x40,0x40,0x40,0x40,0x7C],#L
    [0x81,0xC3,0xA5,0x99,0x81,0x81,0x81,0x81],#M
    [0x00,0x42,0x62,0x52,0x4A,0x46,0x42,0x00],#N
    [0x3C,0x42,0x42,0x42,0x42,0x42,0x42,0x3C],#O
    [0x3C,0x22,0x22,0x22,0x3C,0x20,0x20,0x20],#P
    [0x1C,0x22,0x22,0x22,0x22,0x26,0x22,0x1D],#Q
    [0x3C,0x22,0x22,0x22,0x3C,0x24,0x22,0x21],#R
    [0x00,0x1E,0x20,0x20,0x3E,0x02,0x02,0x3C],#S
    [0x00,0x3E,0x08,0x08,0x08,0x08,0x08,0x08],#T
    [0x42,0x42,0x42,0x42,0x42,0x42,0x22,0x1C],#U
    [0x42,0x42,0x42,0x42,0x42,0x42,0x24,0x18],#V
    [0x00,0x49,0x49,0x49,0x49,0x2A,0x1C,0x00],#W
    [0x00,0x41,0x22,0x14,0x08,0x14,0x22,0x41],#X
    [0x41,0x22,0x14,0x08,0x08,0x08,0x08,0x08],#Y
    [0x00,0x7F,0x02,0x04,0x08,0x10,0x20,0x7F],#Z
    [0x08,0x7F,0x49,0x49,0x7F,0x08,0x08,0x08],#中
    [0xFE,0xBA,0x92,0xBA,0x92,0x9A,0xBA,0xFE],#国
]

**hspi.DEV_HARDWARE_SPI_begin("/dev/spidev0.0")**  #<- Problem is here
hspi.DEV_HARDWARE_SPI_ChipSelect(3)

class MAX7219:
    def __init__(self):
        self.cs_pin = 8
        GPIO.setmode(GPIO.BCM)
        GPIO.setwarnings(False)
        GPIO.setup(self.cs_pin, GPIO.OUT)
        
    def WriteByte(self, Reg):
        GPIO.output(self.cs_pin, 0)
        hspi.DEV_SPI_WriteByte(Reg)
        
    def Write(self, address1, dat1, address2, dat2):
        GPIO.output(self.cs_pin, 0)
        self.WriteByte(address1)
        self.WriteByte(dat1)
        self.WriteByte(address2)
        self.WriteByte(dat2)
        GPIO.output(self.cs_pin, 1)
        
    def Init(self):
        self.Write(0x09,0x00,0x09,0x00)
        self.Write(0x0a,0x03,0x0a,0x03)
        self.Write(0x0b,0x07,0x0b,0x07)
        self.Write(0x0c,0x01,0x0c,0x01)
        self.Write(0x0f,0x00,0x0f,0x00)

led = MAX7219()
led.Init()

while True:
    for j in range(0, 37):
        for i in range(1, 9):
            led.Write(i, disp[j+1][i-1], i, disp[j][i-1])
        time.sleep(1)

aivarannamaa, 2020-06-12:
From the shebang line it looks like this script is meant to be run with Python 3, but Thonny uses Python 3. 

Try running the script from command line with `python3` and let me know whether it works or not.

rofi2809, 2020-06-13:
with python 3 the same error occurs also in terminal, you are right. Is there an alternative command for python 3 or what would be a good solution?

aivarannamaa, 2020-06-13:
If your program works with Python 2 then you could use an IDE which supports it (eg Geany). Or you could approach the producer of the display for advice. Or ask about your problem in Raspberry forum (https://www.raspberrypi.org/forums/)

rofi2809, 2020-06-13:
Thank you very much! ive managed to rewrite the pyhton2 depending codeparts with spidev,  working with pyhton3

closed 2020-06-13 by rofi2809

krzysztofwasala, 2021-08-18:
Could you share rewritten parts? I have the same issue. Sorry fot digging old topic but I couldn't find nothing more on this.

samwong0127, 2022-02-11:
> Thank you very much! ive managed to rewrite the pyhton2 depending codeparts with spidev, working with pyhton3

Sorry to border, but I am having the same issue. Could you share rewritten parts?