thonny/thonny #1239: The bundled python version doesn't show which installer switches it uses

globalturist, 2020-06-12:
I'm trying to create an automated install for thonny3.2.7 and a separate python3.8.3 that has the same installer switches as the bundled python. However, I cant find any unattend.xml file in the thonny installer nor any information on the github page that would specify which installer switches are used.

Also, if there are any installer switches I can omit I would greatly appreciate it if anyone can point them out because I would like to make the python install as minimal as possible (note that i still need the full python version so embeddable python is not the solution for me).

What I mean by installer switches: [https://docs.python.org/3/using/windows.html](url) section 3.1.3 installing without ui.

Targeted use: Programming ESP32 boards with micropython.

aivarannamaa, 2020-06-12:
It never occurred to me to use silent/unattended Python install while building Thonny's windows bundle, but it looks like the right way to do it!

Currently I'm just installing Python as regular user to a fresh directory via installer GUI and then copy it to Thonny's build directory.

stinos, 2020-06-13:
Slightly relaed: in create_installer.bat it looks like CPython's LICENSE and other files are removed 'to avoid confusion'. I'm not sure that is actually allowed by the license. But even if it were, better would be to rename those file to e.g. PYTHON_LICENSE etc.

aivarannamaa, 2020-06-13:
You're right, Python license should remain. Thanks for noticing this!

closed 2021-02-18 by aivarannamaa