thonny/thonny #1241: Re-/start Thonny

grektor, 2020-06-14:
I have a default Thonny software on my Raspberry 4 (bought 1 month ago) running under Python 3.7.3, however on the device Python 2.7.16 is also available.
I wanted to install few packages - e.g.: pandas and socketio in Terminal - but some of packages recognized (the default) python 2.7.16 as show-stopper (expecting upgraded Python).

So, I changed the priorities of python packages from python 2.7.16 to python 3.7.3 to enable installation. Installations (theoretically) are completed perfectly (without error messages).

After the reboot of my raspberry pi, Thonny has not started (neither "mu" nor some basic python games). I changed the priorities back from 3.7.3 to 2.7.16, but then only the "basic python games" have been cured. Thonny has not started. Note: before the appearance of the issue (in the shell of Thonny, python 3.7.3 was displayed).

I tried to run it via Terminal with the following feedback:

pi@raspberrypi:/usr/bin $ ./thonny
Traceback (most recent call last):
  File "./thonny", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/pi/.local/lib/python3.7/site-packages/pkg_resources.py", line 1479, in <module>
    register_loader_type(importlib_bootstrap.SourceFileLoader, DefaultProvider)
AttributeError: module 'importlib._bootstrap' has no attribute 'SourceFileLoader'

I have searched on "pkg_resources.py", line 1479" to find similar recorded (and hopefully solved issues), but without finding explicit solution. In all documented cases the issue was around basic installation of python 3.x which is not in my case.

Any suggestion? Thank you!

aivarannamaa, 2020-06-15:
> So, I changed the priorities of python packages from python 2.7.16 to python 3.7.3 to enable installation.

How did you do this? 

Python 2 and Python 3 are supposed to be usable side-by-side, you just use different commands with them (eg. `python` and `pip` vs `python3` and `pip3`). If you want to install pandas with `apt` then you should `apt install python-pandas` for Python 2 and `apt install python3-pandas` for Python 3.

grektor, 2020-06-15:
Thank you for your suggestion!

Either I use "pip" or "pip3" to install "socketio", both commands uninstall "setuptools" 47.1.1 and re-install the version with 3.3 - automatically (see below). Of course the socketio==0.2.1 is available after both installations, but Thonny stops functioning!
I re-installed "setuptools"==47.1.1, restart Thonny and tried to run python script (with socketio) which gave the message: NameError: name 'SocketIO' is not defined
-------------------------------------------
pi@raspberrypi:~ $ pip3 install socketio
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: socketio in ./.local/lib/python3.7/site-packages (0.2.1)
Collecting setuptools==3.3
  Using cached setuptools-3.3-py2.py3-none-any.whl (545 kB)
Requirement already satisfied: netifaces==0.10.6 in ./.local/lib/python3.7/site-packages (from socketio) (0.10.6)
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 47.1.1
    Uninstalling setuptools-47.1.1:
      Successfully uninstalled setuptools-47.1.1
Successfully installed setuptools-3.3

pi@raspberrypi:~ $ pip install socketio
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: socketio in ./.local/lib/python3.7/site-packages (0.2.1)
Requirement already satisfied: netifaces==0.10.6 in ./.local/lib/python3.7/site-packages (from socketio) (0.10.6)
Collecting setuptools==3.3
  Using cached setuptools-3.3-py2.py3-none-any.whl (545 kB)
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 47.2.0
    Uninstalling setuptools-47.2.0:
      Successfully uninstalled setuptools-47.2.0
Successfully installed setuptools-3.3
--------------------------------------------------------
Concerning PANDAS, I have got new error messages. According to the link below, I need to use "sudo" however It is not recommended, since I can overwrite accidentally something which can cause bigger chaos.
https://askubuntu.com/questions/223484/permission-denied-are-you-root

pi@raspberrypi:~ $ apt install python-pandas
E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?
pi@raspberrypi:~ $ apt install python3-pandas
E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?

Any recommendation?

aivarannamaa, 2020-06-16:
To clear the picture a bit, you may want to remove user-installed packages (probably under `~/.local/lib/python3.7/site-packages`). This should be rather safe operation as system utilities don't depend on them.

This may already help.

Regarding socketio requiring exactly setuptools 3.3 -- this is rather unusual. Setuptools is so widely used package that requiring an exact version of it does not make sense.

I recommend you forget socketio for now and try to get your Python working again. Once you do and want to work in Thonny, create a new virtual environment (Run => Select interpreter => Alternative Python 3... => Create new virtual) and try to install socketio there (Tools => Manage packages or Tools => Open system shell). This way you don't mess up the rest of the system.

closed 2020-06-19 by aivarannamaa