thonny/thonny #1243: some of the shell functions dont work

Ochiento, 2020-06-16:
i cannot run or debug programs from thonny on my raspberry pi

aivarannamaa, 2020-06-16:
I need more details in order to help you.
Why do you think some of the shell functions don't work?
Do you get an error message?

Ochiento, 2020-06-16:
the run current script, debug current script, step over and step into, step
out and resume buttons dont work.

aivarannamaa, 2020-06-16:
Do you mean the buttons are grayed out / inactive?
Do you have menu bar available? If yes then please select "Run => select interpreter" and tell me which back-end is selected (or send me a screenshot). 

You could also send me all non-empty files from `~/.config/Thonny`.

Ochiento, 2020-06-16:
what do you mean by backend which   item in the menu bar on the second pic
i should click

aivarannamaa, 2020-06-16:
> what do you mean by backend

I meant the value in the combobox under "Which interpreter or device should Thonny use for running your code?"

I don't see the pic. GitHub has problems with attachments when sent via e-mail.

Ochiento, 2020-06-16:
the same interpret which runs thonny

aivarannamaa, 2020-06-17:
Please open Terminal and run `thonny`. Do you see any error messages in the Terminal?

Ochiento, 2020-06-17:
Tried running it I got delegated to an existing thonny instance. Exiting now

Ochiento, 2020-06-17:
Tried running thonny in terminal and I got the message "Delegated to an
existing Thonny instance. Exiting now.

aivarannamaa, 2020-06-17:
Close Thonny window first before doing this

Ochiento, 2020-06-17:
I did it, it opened and the button is still greyed out

aivarannamaa, 2020-06-17:
No error messages in Terminal? Can you send me the screenshot in GitHub?

Ochiento, 2020-06-17:
[image]
there is a pic, not that good looking

Ochiento, 2020-06-17:
I sent it

aivarannamaa, 2020-06-17:
Oh, I should have said that I want whole Thonny window :)
I wanted to see whether you have anything in Shell. Can you enter commands in Shell? Is the "Run current script" command disabled in "Run" menu as well

Ochiento, 2020-06-17:
[image]

there it is

aivarannamaa, 2020-06-17:
This is important information that the shell shows. Please copy all of it here. 

(In the future, if you want technical help from someone, then provide as much information as you can. Whenever you see something that looks like an error message then forward all of it. In this case you should have started with this error.)

Ochiento, 2020-06-17:
I tried using stop/restart but the script buttons light up and then grays out again

aivarannamaa, 2020-06-17:
> I tried using stop/restart but the script buttons light up and then grays out again

I understand this, but I need to get the whole error message that you see in the Shell in order to help you.

Ochiento, 2020-06-17:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/parso/cache.py", line 12, in <module>
    import cPickle as pickle
ModuleNotFoundError: No module named 'cPickle'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/backend_launcher.py", line 76, in <module>
    VM().mainloop()
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 122, in __init__
    self._load_shared_modules()
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 298, in _load_shared_modules
    self.load_modules_with_frontend_path(["parso", "jedi", "thonnycontrib", "six", "asttokens"])
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 308, in load_modules_with_frontend_path
    import_module(name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/parso/__init__.py", line 42, in <module>
    from parso.grammar import Grammar, load_grammar
  File "/usr/lib/python3/dist-packages/parso/grammar.py", line 10, in <module>
    from parso.cache import parser_cache, load_module, save_module
  File "/usr/lib/python3/dist-packages/parso/cache.py", line 14, in <module>
    import pickle
  File "/home/pi/programs/pickle.py", line 4, in <module>
    pickle.dump(mylist, f)
AttributeError: module 'pickle' has no attribute 'dump'

Backend terminated or disconnected. Use 'Stop/Restart' to restart ...

aivarannamaa, 2020-06-17:
It looks like you have a file named pickle.py in your working directory, which shadows a library module with same name. Try renaming it.

Ochiento, 2020-06-17:
you are incredible, thank you so much

closed 2021-02-18 by aivarannamaa