thonny/thonny #1244: Editor RMB

DenToetie, 2020-06-19:
OS = Win7
- English= "Comment out" must be "Comment"
- mnemonic key must be AltGr+3 and not Alt+3 (Alt+3 set the line in comment but also gives a hart in the text : # § random index♥ )

aivarannamaa, 2020-07-20:
Thank you for the report and sorry for late reply!

> English= "Comment out" must be "Comment"

I took "Comment out" from IDLE. Are you sure "Comment" would be better? If I were a beginner with piece of text is selected in the editor, I would expect "Comment" to add a comment to the selected text, not make the selected text a comment.

> mnemonic key must be AltGr+3 and not Alt+3 (Alt+3 set the line in comment but also gives a hart in the text : # § random indexhearts )

Can't confirm. This probably depends from the keyboard layout. Which one are you using? I can perform this command with Alt+3 both in Windows 7 and in Ubuntu, both with Estonian and US layout. AltGr+3 deletes the selected text and replaces it with `£` with Estonian layout and does the same as Alt+3 with US layout.

DenToetie, 2020-07-28:
Q1 = Comment
Or maybe=  "Comment out" must be "Make Comment" ?

Q2 = mnemonic key for "Comment out"
I'm working on a laptop HP Elitebook 8740w with Win7 SP1 Eng
Keyboard driver = Standard 101/102-Key or Microsoft Natural PS/2 Keyboard
with HP QLB
Keyboard layout name= Belgian (period)

I hope this will help you.
mvg,

Guido Geurs
Gaverbosdreef 28
9630 Zwalm
Tel: 055-61.67.74
E-mail: guido.geurs@gmail.com

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

aivarannamaa, 2022-04-01:
I'll leave these as is for now, because different IDE-s doesn't seem to agree on which shortcut to use for adding / removing comments. In 4.1 I hope to implement the possibility to assign custom shortcuts (#306) and then I'll remove these default shortcuts altogether.

closed 2022-04-01 by aivarannamaa