thonny/thonny #1247: Crash on loading uPython on esp32

varna9000, 2020-06-21:
Hi, I've got Mac OS Catalina and tried Thonny 3.2.6 and 3.2.7 pkg and dmg installs. Both crash when trying to load micropython 1.11 on esp32 via FTDI adapter. Using regular serial monitor board loads ok. The crash is the following:
```
Crash in backend
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 103, in __init__
    self._prepare(clean)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 120, in _prepare
    self._builtin_modules = self._fetch_builtin_modules()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 174, in _fetch_builtin_modules
    assert not err, "Error was: %r" % err
AssertionError: Error was: 'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nNameError: name \'help\' isn\'t defined\r\n'
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 103, in __init__
    self._prepare(clean)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 120, in _prepare
    self._builtin_modules = self._fetch_builtin_modules()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 174, in _fetch_builtin_modules
    assert not err, "Error was: %r" % err
AssertionError: Error was: 'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nNameError: name \'help\' isn\'t defined\r\n'

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
```

aivarannamaa, 2020-06-23:
Can you please try following command on your device's REPL using another tool:

```
help("modules")
```

Does it work?

varna9000, 2020-06-23:
@aivarannamaa no, it doesn't. Output is:

`>>> help('modules')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'help' isn't defined`

aivarannamaa, 2020-06-23:
Thanks for the information! I thought that `help` is missing only from the Unix port. I'll take this into account. 

BTW, from where did you download your firmware?

varna9000, 2020-06-23:
@aivarannamaa I got it from here. I think the author removed the help and few other modules to first the camera module and save some memory. [https://github.com/shariltumin/esp32-cam-micropython/tree/master/firmwares](url)

aivarannamaa, 2020-10-28:
Fixed in 3.3b7

closed 2020-10-28 by aivarannamaa