thonny/thonny #1248: Thonny fails to run in a Linux Environment (workon <env>)

jenapk, 2020-06-21:
Hi,
I use Thonny in my Raspberry Pi 4 and facing the following issue. 

How to run THONNY in a specific linux environment ? I want say about the linux command  "workon <environment name>". 
A python program depends on the environment and not runs without it. I use this scenario in face-recognition application. So, I use text editor for .py modification.

Please suggest how to use thonny in this case.

regards
PK Jena
+91 9381708220
pkjena@gmail.com

aivarannamaa, 2020-06-23:
I'm not sure whether I understood your problem correctly.

Do you want to set specific environment values for your program? Currently Thonny does not support this, but your program will inherit all environment variables from Thonny GUI process, so you could try following:

`MY_ENV_VAR=my_value thonny`

jenapk, 2020-06-23:
Hi,
Thank you for your early response.
I failed to explain you properly. My issue as below...may be little lengthy to explain..

I am running an application based on OpenCV in Pi4.  It is working and each time I execute the python script from a 64-bits linux machine (my laptop) and transfer to Pi4. So I tried to do same with thonny. 

In Pi4, my python program does not run directly (python face.py), because OpenCV module is  compiled and installed  in an environment (I named it as "cv"), So I execute manually "workon cv" at command prompt and then execute the python script.

$ workon cv    
(cv) $python face.py

Now mu code works fine. Same I want to do with Thonny.
After executing "workon cv", a prefix of "cv" to linux command appears. This is the environment in linux. 
is is nothing about any environmental variable.
 
I hope, you understood, otherswise, kindly let me know.

regards
PK JENA
+9381708220
pkjena@gmail.com

aivarannamaa, 2020-06-23:
I'm not familiar with `workon` command, but if it has to do with a Python virtual environment, then you may choose its executable as Python's backend:

* run `python` in your environment and then find out the location of its executable (`import sys; print(sys.executable)`)
* Thonny => Run => Select interpreter => Custom Python 3 ... => Locate another Python

After this Thonny should run your programs in that virtual environment.

thelwyn, 2020-09-20:
> I'm not familiar with `workon` command, but if it has to do with a Python virtual environment, then you may choose its executable as Python's backend:
>
>     * run `python` in your environment and [...]

An issue in my case is the my Python virtual environment is stored in an hidden directory .virtualenvs and Thonny GUI doesn't permit to select an hidden folder. Is there a conf file somewhere that I could manually edit to add the path instead of using the GUI?

jenapk, 2020-09-20:
Thank you so much for your reply.
I saw the option in Thonny, but I use in the py script instead. It works
fine for me as of now.

I appreciate your level of support as well as your concern.

Thank you again.

Best regards
PK Jena
India.

aivarannamaa, 2020-09-20:
> An issue in my case is the my Python virtual environment is stored in an hidden directory .virtualenvs and Thonny GUI doesn't permit to select an hidden folder. Is there a conf file somewhere that I could manually edit to add the path instead of using the GUI?

I'll include hidden directories in the future. For now you can 

* open Thonny configuration directory (Tools => Open Thonny data folder)
* close Thonny
* edit `path` in `CustomInterpreter` section, eg:

```
[CustomInterpreter]
path = /usr/bin/python3
```

* save
* start Thonny

aivarannamaa, 2020-09-20:
@thelwyn, which platform are you using? Can't you switch on hidden files when you right-click in the directory selection dialog?

thelwyn, 2020-09-20:
> @thelwyn, which platform are you using? Can't you switch on hidden files when you right-click in the file selection dialog?

Rasputin Buster

jenapk, 2020-09-20:
Hi.

I am working on raspberry Pi4.
Definitely will do it.
Allow me sometime.

Regards
PK Jena

thelwyn, 2020-09-21:
> 
> 
> @thelwyn, which platform are you using? Can't you switch on hidden files when you right-click in the file selection dialog?

Yes, I can indeed show the hidden files by right-clicking in the dialog window! Sorry I didn't know this possible action, that's not very intuitive nor common in file browser windows. So no issue in tend, or an usability one. Thanks for your fast help!

closed 2021-02-18 by aivarannamaa