thonny/thonny #1249: from random import shuffle function

barakat99, 2020-06-25:
Bonjour je n'arrive pas à utiliser la fonction from random import shuffle pour remuer les élements de ma liste en fait dans l'interpréteur shell je fais comme çeci:
>>> from random import shuffle
>>> l1 = [1, 2, 3, 4]
mais rien se met en désordre.
Veuillez me résoudre ce problème s'il vous plaît.
Cordialement.
Mostafa Barakat.

barakat99, 2020-06-25:
Bonjour je n'arrive pas à utiliser la fonction from random import shuffle pour remuer les élements de ma liste en fait dans l'interpréteur shell je fais comme çeci:

            from random import shuffle
            l1 = [1, 2, 3, 4]
            mais rien se met en désordre.
            Veuillez me résoudre ce problème s'il vous plaît.
            Cordialement.
            Mostafa Barakat.

aivarannamaa, 2020-06-25:
Sorry, I don't speak French, but it looks like your question is not related to Thonny, so I recommend you ask it in a general Python forum.

closed 2020-06-25 by aivarannamaa