thonny/thonny #1264: to use a microbit

PKHG, 2020-07-13:
I would like ut use the microbit using thonny (w10)
How to do this for thonny:
 I need 
pip install cs20-microbitio   installed by thonny ;-)

I have (sorry) several pip.exe  but hoe to make it (if done) it available for thonny
Greetings
Peter

aivarannamaa, 2020-07-14:
If you want to use it in Thonny, then go to "Tools => Manage packages", enter `cs20-microbitio` to the search box and install it from there.

PKHG, 2020-07-14:
Thanks, finally found the  install option and it looks good, BUT trying it via:
import microbit delivers:
>>> import microbit
Using path:C:\Users\Zbook\AppData\Roaming\Python\Python37\site-packages\microbit
Yay, I loaded the BITIO packaged pyserial
will reuse cache or scan for new port
Using port:COM4
connecting...
getting active serial port connection to micro:bit
creating a raw REPL connection via serial
Traceback (most recent call last):
  File "<pyshell>", line 1, in <module>
  File "C:\Users\Zbook\AppData\Roaming\Python\Python37\site-packages\microbit\__init__.py", line 127, in <module>
    repl = repl.REPL(s)
AttributeError: module 'repl' has no attribute 'REPL'

So, what needs to be done? 

Please let me know ;-)

aivarannamaa, 2020-07-14:
I don't know, this package is not related to Thonny.

If you really need this package (ie. control your micro:bit from your PC, eg. for controlling Minecraft), then you need to contact its author. If you just want to program micro:bit, then 

* read about Thonny's MicroPython support: https://github.com/thonny/thonny/wiki
* plug in your micro:bit
* select micro:bit backend (Run => Select interpreter => MicroPython (micro:bit)) 
* install firmware ("Open the dialog for installing ..." button)
* press "OK"

closed 2020-10-15 by aivarannamaa