thonny/thonny #1266: Backend terminated or disconnected. Use 'Stop/Restart' to restart ...

Rene498, 2020-07-14:
i tried Stop/restart but it isnt working.  i cant run any program i write. i have this in my shell
can you help me please

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/backend_launcher.py", line 76, in <module>
    VM().mainloop()
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 122, in __init__
    self._load_shared_modules()
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 298, in _load_shared_modules
    self.load_modules_with_frontend_path(["parso", "jedi", "thonnycontrib", "six", "asttokens"])
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 308, in load_modules_with_frontend_path
    import_module(name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/jedi/__init__.py", line 41, in <module>
    from jedi.api import Script, Interpreter, set_debug_function, \
  File "/usr/lib/python3/dist-packages/jedi/api/__init__.py", line 29, in <module>
    from jedi.api.project import get_default_project
  File "/usr/lib/python3/dist-packages/jedi/api/project.py", line 2, in <module>
    import json
  File "/usr/lib/python2.7/dist-packages/flask/json/__init__.py", line 13, in <module>
    from flask.globals import current_app, request
  File "/usr/lib/python3/dist-packages/flask/__init__.py", line 19, in <module>
    from jinja2 import Markup, escape
  File "/usr/lib/python3/dist-packages/jinja2/__init__.py", line 33, in <module>
    from jinja2.environment import Environment, Template
  File "/usr/lib/python3/dist-packages/jinja2/environment.py", line 15, in <module>
    from jinja2 import nodes
  File "/usr/lib/python3/dist-packages/jinja2/nodes.py", line 19, in <module>
    from jinja2.utils import Markup
  File "/usr/lib/python3/dist-packages/jinja2/utils.py", line 41, in <module>
    _slash_escape = '\\/' not in json.dumps('/')
AttributeError: module 'json' has no attribute 'dumps'

Backend terminated or disconnected. Use 'Stop/Restart' to restart ...

────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/backend_launcher.py", line 76, in <module>
    VM().mainloop()
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 122, in __init__
    self._load_shared_modules()
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 298, in _load_shared_modules
    self.load_modules_with_frontend_path(["parso", "jedi", "thonnycontrib", "six", "asttokens"])
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 308, in load_modules_with_frontend_path
    import_module(name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/jedi/__init__.py", line 41, in <module>
    from jedi.api import Script, Interpreter, set_debug_function, \
  File "/usr/lib/python3/dist-packages/jedi/api/__init__.py", line 29, in <module>
    from jedi.api.project import get_default_project
  File "/usr/lib/python3/dist-packages/jedi/api/project.py", line 2, in <module>
    import json
  File "/usr/lib/python2.7/dist-packages/flask/json/__init__.py", line 13, in <module>
    from flask.globals import current_app, request
  File "/usr/lib/python3/dist-packages/flask/__init__.py", line 19, in <module>
    from jinja2 import Markup, escape
  File "/usr/lib/python3/dist-packages/jinja2/__init__.py", line 33, in <module>
    from jinja2.environment import Environment, Template
  File "/usr/lib/python3/dist-packages/jinja2/environment.py", line 15, in <module>
    from jinja2 import nodes
  File "/usr/lib/python3/dist-packages/jinja2/nodes.py", line 19, in <module>
    from jinja2.utils import Markup
  File "/usr/lib/python3/dist-packages/jinja2/utils.py", line 41, in <module>
    _slash_escape = '\\/' not in json.dumps('/')
AttributeError: module 'json' has no attribute 'dumps'

Backend terminated or disconnected. Use 'Stop/Restart' to restart ...

aivarannamaa, 2020-07-14:
If you have your script named `json.py` then rename it -- it is shadowing a system module

aivarannamaa, 2020-07-14:
I was mistaken, it looks like for some reason you have `/usr/lib/python2.7/dist-packages/flask/` in your Python path and this confuses Thonny. 

Do you have this directory as your current directory?

Rene498, 2020-07-15:
Yes, how do i change it? I cant find anything

aivarannamaa, 2020-07-15:
Thonny changes automatically to a directory if you run a script from it. You can change manually by opening "View => Files" and navigating to required folder (double clicking on a directory brings it into focus and this equals to making it a current directory). If you don't want to change the current directory automatically then uncheck Tools => Options => Run & Debug => Change ...

Do you need to run a script from `/usr/lib/python2.7/dist-packages/flask/` ? If yes, then it can be problem even if you change your working directory as Python adds script directory to path.

closed 2021-02-18 by aivarannamaa