thonny/thonny #1268: Add "NOT use Help" checkbox in Tools-options-somewhere

WSDsuperace, 2020-07-15:
I have used Thonny a long time and yesterday installed a new pc and upgraded to the newest Thonny.
I use Thonny  (now 3.2.7) for writing Micropython for Esp32 and Esp8266 modules. I run all code on the boards. As I am sure you all know these modules have limited recourses compared to a regular PC. To minimize the size of the custom made firmware (firmware.bin) I regularly do not include the Help module when I freeze the python files. So?

It seems that Thonny handles the Help module as a mandatory module to be able to even start the GUI. A board using a micropython firmware with no help module will result in a hard crash when starting Thonny. Of course not a big surprise because the \plug-ins\micropython\backend.py is trying to access the help module.

Workaround: Comment out most of the content of  \plug-ins\micropython\backend.py  "def _fetch_builtin_modules(self):" but that really is an ugly hack.

Can we get at checkbox? So Thonny has two startup "modes", the normal one and "You better be sure what you are doing-mode"

On a different note, now while I am writing.... Can we please get 3 new icons,  "Save to MAIN on Device" and "Save to Boot on Device" and "Save using current name on Device". I am sorry but the new way (Save as --> ) is so much more trouble than the old way. For embedded software development Save to Main ,Boot and Current name is something done a huge number of times every day. Saving seconds and clicks are actually important here.  OR keep the 3 old ways under Device menu while the new way also will work.          

At least my two cents....

Anyways, thanks for a great software.

aivarannamaa, 2020-07-16:
Thanks for your input!

Missing `help` module has been reported already (#1247). In the next release this will be solved without requiring a checkbox. (I do intend to provide means for advanced configuration, but for other reasons: #970.)

I'll rename this issue to cover your second suggestion only (in the future I recommend to open separate issues for separate topics).

> On a different note, now while I am writing.... Can we please get 3 new icons, "Save to MAIN on Device" and "Save to Boot on Device" and "Save using current name on Device". I am sorry but the new [...]

I'm not doing embedded software development myself, so my vision may have been impractical indeed. Actually, while refactoring MicroPython related UI, I was focusing on beginners playing with BBC micro:bit or something like this. Therefore I optimized for writing the scripts directly on the device.

I still want to keep focus on beginners, but I realize that some more advanced users (attracted mostly by MicroPython support) are also good for the project, as they tend to be more demanding and write better issues :)

I assume most serious programmers want to write their code on a hard drive and copy it to the device later. In this case, "Save copy as" would be more suitable than "Save as". There have been proposals about automatic sync (#531, #990) but it is not easy to implement a foolproof and simple to use sync.

What do you think of this more general feature: #1270 ? After exporting your local main.py and boot.py once you would be able to export them again with pressing Export shortcut (eg. Ctrl+E) and ENTER.

aivarannamaa, 2020-07-16:
The solution involvin Export command would relieve me of worrying about different main scripts: 

* CircuitPython allows `main.py`, `code.oy`, `main.txt`, `code.txt` with different priorities)
* On some MicroPython devices it's `/main.py` on others it's `/flash/main.py`

aivarannamaa, 2020-07-16:
... and it's more universal -- you could use it for easily updating your helper modules as well.

WSDsuperace, 2020-07-16:
I have done software development now fulltime for almost 20 years using different platforms. How about I explain my point of view?  One project will be split on many files, some are pure libraries that are later imported as modules, some are code that will run depending on different cases. So I may have 5-6 files open at the same time. 

Because I write for embedded I want to test my codes and changes on the actual hardware, a PC as I said before does not behave the same way as the real hardware. There are so many different IDE on the market but what I love about Thonny is the easy way to move and upload files (well it was easy) :) You are right that I store the code on the PC, but I test the Code on the PCB. I could make a change in any file, Device Upload using current name , wait a few seconds for the loading to finish. Hit PCB reset button and watch the results. Do some changes, and repeat, and repeat. You get the idea. Because I use Upload using current name I don't have to care at all what the name is, Thonny just uploads the selected file and done and done. 

I totally understand you want to make it easy for beginners. Easy is always good.  The old way was very clear. For Main click here, For Boot click here and for others click here.....  I used Thonny in a summer camp last year and it took less than 2 minutes for the kids to understand the Boot, Main and Other concept of loading.

I can see benefits with the new way too but it was the old way that made Thonny stick out from all others IDE.

aivarannamaa, 2020-07-16:
> The old way was very clear. For Main click here, For Boot click here and for others click here..... I used Thonny in a summer camp last year and it took less than 2 minutes for the kids to understand the Boot, Main and Other concept of loading.

On the other hand "Save copy as" would be more general and transparent (less specific or magical). Also a bit more cumbersome, agreed.

I'm definitely not saying that restoring (some of the) old commands is out of the question. I just want to consider as many alternatives as possible.

I suppose for experienced users the most convenient solution would be a command with a shortcut, which automatically writes changed files to the device and performs a soft reboot, ie. runs the code. As I said, syncing is not easy but do you think it is worth considering?

A related issue -- I've seen two different and conflicting terms used for the act of copying something from PC to a device: "uploading" and "downloading". I was explained that it depends on the viewpoint. What's your opinion -- could the command "Upload" be confusing for some users?

WSDsuperace, 2020-07-17:
Personally, automatic writes and a soft reboot is not a feature I would like. Why? I want/need to decide when the code starts running (mostly by either power on or RST button) and prefer that the IDE only does what is asked to do (load file). Please  no autosync :)

Yes words, all people have their own personal feelings towards them. These are mine:
"Save" I save a file so it is stored and can be accessed again 

"Save as" Here I want to keep the content just change the name so it can be accessed and edited with the new name

"Download" I would use this to describe the process of getting a file to my PC for example: Download the file and Save it in C:\temp

"Upload"  The process of sending a file somewhere for storage so it may be accessed later.

All these words implies for me that I will access the file again at a later time.
I normally use "Load or Put" when sending files to a PCB, I will not access those files again, the CPU will but I will not . In many cases it may not even be possible to get them out again :)
'Download the file --> Save it locally --> Load the file to the PCB'
But the short answer is "NO" I do not think a user is confused by "Upload", even more clear is "Upload to Device"

I am back to 3.1.2 so I have "the old style" again :) with a hacked backend.

You brought up "Convenient", what do you think about this? 
Batch Uploading
For example:
We create a new file similar to a makefile or dockerfile, an ordinary text file. Each row in this file has a filename we want to Upload to device. By selecting this function we load all files at one time.
What are the benefits?
A Github project prepared with this special file can be downloaded as Zip, extracted and then easily loaded to the Device. It would be very easy to try different examples even when they span several files. You could probably look at it as very simple project management. I feel it would be super good for teachers and similar. An easy way to make sure all boards are loaded the same way. 

Many IOT projects have html and config files that must be loaded on the device too and including their names in the new file would make 1 click loading possible.

What is your feeling about batch uploading?

aivarannamaa, 2020-07-17:
> We create a new file similar to a makefile or dockerfile, an ordinary text file. Each row in this file has a filename we want to Upload to device.

I like the directness of this approach. Maybe use [MANIFEST.in](https://packaging.python.org/guides/using-manifest-in/) files for this?

aivarannamaa, 2020-07-17:
> Maybe use MANIFEST.in files for this?

... or rather not -- many files in a source distribution may be unnecessary on the device.

The most obvious existing model would be setup.py, but I don't know how to express the intention that some modules should go to the root directory and some to `/lib`.

I started a discussion in MP forum: https://forum.micropython.org/viewtopic.php?f=2&t=8736