thonny/thonny #1269: Crash when trying to use camera on macOS 10.15

wallysalami, 2020-07-15:
If I try to run this simple code on Thonny on macOS 10.15

```python3
from cv2 import *
stream = VideoCapture(0)
```

I get this crash 

```
Process:               Python [25430]
Path:                  /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               3.7.7 (3.7.7)
Code Type:             X86-64 (Native)
Parent Process:        Python [25423]
Responsible:           Python [25423]
User ID:               501

Date/Time:             2020-07-15 20:49:39.424 -0300
OS Version:            Mac OS X 10.15.4 (19E287)
Report Version:        12
Anonymous UUID:        EF27202F-4B84-5E51-581E-88F299DEE851

Time Awake Since Boot: 110000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Abort trap: 6
Termination Reason:    Namespace SIGNAL, Code 0x6
Terminating Process:   Python [25430]
```

But if I ran the same code in Terminal, I get a dialog window requesting camera access

[image]

I think Thonny is not requesting Camera and Microphone permissions in macOS.

aivarannamaa, 2020-07-17:
Do you have latest Thonny (3.2.7) installed with a pkg installer?

Thonny used to have similar permission issues with accessing Desktop and Documents directories, but I hoped these were solved with 3.2.7 pkg installer.

Do you get same problem when you start your script from Thonny but with "Run => Run current script in Terminal" ?

wallysalami, 2020-07-17:
The version was already 3.2.7, but I reinstalled with the pkg just to be sure. The problem is still there.

The code runs with no issues if I use that "Run => Run current script in Terminal".

aivarannamaa, 2020-07-17:
At the moment I don't have any ideas how to solve it but I'll keep it in the TODO list.

aivarannamaa, 2021-02-27:
Please try whether https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg works better for you. It's with an experimental Universal 2 variant of Python 3.9.2.

wallysalami, 2021-04-01:
Hi again!

Sorry for not getting back to you sooner. I was having some trouble installing and running OpenCV with Python 3.9 in macOS Big Sur.

So today I tried to capture an image from the webcam calling `ffmpeg` via Python's `os.system`. The result was the same:  it only works with "Run => Run current script in Terminal".

mithos91, 2024-06-12:
After 3 years, still same issue. Can't authorize camera from thonny ence only way to use it is by opening via terminal. It is so annoying.